UbiquityRobotics/fiducials #1: Auto detect height of first fiducial in map

jim-v, 2016-01-06:


rohbotics, 2017-01-03:
We want to be able to have a auto_init_map, which takes an initial image from a ros topic.
It should initialize the the current `/base_link` position to zero, and place all currently visible fiducials in the map, then exit.

jim-v, 2017-01-06:
The code to do this is in updateMapFromExternal().  We could have fiducial_slam_node automatically initialize the map if there isn't one already, or maybe have a service call to do this, so it is less mysterious to the user.

waynegramlich, 2017-01-07:
All:

My $.02 on this stuff is that you want to do as much automagically as possible.

After that we want error messages that both describe the problem *AND* suggest
the solution:

* No camera connected to topic /...
* Fiducial sizes not specified in /.../foo.yaml.

The goal is that when all the error messages are gone, it is working.

Regards,

-Wayne

closed 2017-01-12 by jim-v