UbiquityRobotics/fiducials #11: Basic Automated Tests

rohbotics, 2016-11-23:
We need tests! We can start out with some basic ones.

- [ ] Publish image(s) of a fiducial in frame, and make sure that the detection outputs a sane ROI.
- [ ] Use existing map, publish image of fiducial, check for sane localization.
- [ ] Image with two fiducials, connect them
- [ ] Bag file with video of moving through a bunch of fiducials, map should make sense

jim-v, 2016-11-28:
@rohbotics Is there a place I can host a large bag file?

rohbotics, 2016-11-28:
We should look into a static host how large files, like AWS S3, but for now I think you will have to do it from google drive/dropbox

jim-v, 2016-11-30:
The basic tests are in place.

rohbotics, 2016-11-30:
Great, I will take a look closer look today.

rohbotics, 2016-12-16:
Sorry for the delay, Looks good!

closed 2016-12-16 by rohbotics