UbiquityRobotics/fiducials #45: Create a utility for generating aruco markers

jim-v, 2017-01-12:


closed 2017-01-16 by jim-v

louislim70, 2018-01-01:
When i tested, the aruco_detect package in fiducials could not detect markers created by "http://terpconnect.umd.edu/~jwelsh12/enes100/markergen.html".
But, the marker of the test image(tag_01_d7_14cm.png) in aruco_detect/test/test_images is detected with marker id is 1.

The shape of two images is very different but the dictionary is same as aruco::DICT_5X5_250.

I also tested script (create_markers.py) to make some aruco markers.
I don't know if i want make the marker of aruco::DICT_5X5_250.
So, just typed like "create_marker.py 1 2 marker_pdf_file".
And then, there is a pdf file with 2 markers (8x8)which was not detected  in same directory.

What's problem??

jim-v, 2018-01-01:
Hi,
By default, the dictionary used by `aruco_detect` is 7 (DICT_5X5_1000), but this can be changed with the `dictionary` parameter.  I've updated the (documentation)[ http://wiki.ros.org/aruco_detect] to list all the available dictionaries.

The arguments to `create_markers.py` are `startId, endId, pdfFile, [dictionary]`.   You can use the optional dictionary argument to choose a different one.

If you have any more questions, please create a new issue - this one is closed.
Thanks.

louislim70, 2018-01-01:
I know my mistake...
i used /fiducials/fiducial_lib/script/create_markers.py. (it created 8x8 markers)
When i use /fiducials/aruco_detect/script/create_markers.py, it generated 5x5 markers which can detecte.

rohbotics, 2018-01-01:
@jim-v We may need to put more deprecation warnings on the old style fiducials stuff, or even remove it. It should prevent a whole load of support requests.