UbiquityRobotics/fiducials #64: Use kalman filter to integrate with odometry

jim-v, 2017-04-15:


AnasIbrahim, 2019-01-14:
hello,
I'm trying to achieve this task right now, but maybe a little differently.
my setup have an iRobot create base and a Ceiling camera (so camera detect moving fiducials on the robots not the other way around) and i'm trying to fuse them using robot_localization package. so, i guess my tf setup will simply be:
map ------static_transform_to------> odom ------fused_odom------> base_link

The point i'm struggling with is that aruco_detect pkg only publish tf frames but what i need is PoseWithCovarianceStamped msg, i think the obvious difference is the covariance part.

is there anything in the pkg that can help with this?
can fiducials_slam achieve this and i'm missing the whole point of it?

jim-v, 2019-01-14:
Hello,

Thanks for your message.  The `tf` published by the `aruco_detect` node contain the position of the fiducials relative to the camera.  These are used by the `fiducial_slam` node to generate a map and localize the robot.  The pose of the robot is published by `fiducial_slam` as both a `tf` and a `PoseWithCovarianceStamped` message.

Please let us know how you get on with the Kalman Filter.  If you get it working, it would be nice to have some sample configuration files.

Jim

AnasIbrahim, 2019-01-18:
I got the EKF to work, i will be pushing my changes in this [repo](https://github.com/AnasIbrahim/ceiling_cam_fiducials_odom_ekf).

As my configuration is different from what fiducial_slam does (fixed ceiling camera, fiducials on robots), i ran fiducial_slam to get the tf frames but give it param base_frame=dummy_frame, so it only publish the tf frames (another node convert it to pose) and do nothing more. it would be good if the package can publish tf frames in a separat node and even better if it publish PoseWithCovarianceStamped for each fiducial.