UbiquityRobotics/fiducials #73: Marker height drifts

sevenbitbyte, 2017-05-11:
The tf pose of the detected fiducials if very spatially correct however the visualization markers of the same fiducials has high vertical error.

[image]

jim-v, 2017-07-22:
The tf shows the currently estimated pose of the fiducual relative to the camera, while the green and red cuboids show the position of the fiducials within the map.  By design, the map requires multiple pose estimates to update.  However, there have been some bug fixes, eg #83.  Can you please retry this, and save a bag file of /raspicam_node/image/compressed

rohbotics, 2017-12-16:
Closing due to staleness

closed 2017-12-16 by rohbotics