UbiquityRobotics/fiducials #76: Build Error in fiducial_pose, Fiducial.h is missing

maxinho, 2017-05-17:
I get the following error, when I compile the package with catkin_make:

fatal error: fiducial_msgs/Fiducial.h: No such file or directory

This can be fixed by inserting

add_dependencies(rpp_pose ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})

in fiducial_pose/CMakeList.txt

I am on Ubuntu 16.04, Opencv 3.2.0, ROS Kinetic

rohbotics, 2017-05-17:
Thanks for reporting this issue. 

Since you already have a solution, can you please create a Pull Request to contribute this change.

closed 2017-05-21 by jim-v

AbdelrahmanElsaid, 2021-04-19:
i used this  ` sudo apt-get install ros-kinetic-aruco-detect` to install the needed dependencies required for rr_auto_dock pkg

PR #77 (2017-05-21): Fix deps
Closes #76