UbiquityRobotics/fiducials #81: Get test set of fiducial images w/ varying lighting

davecrawley, 2017-06-03:


rohbotics, 2017-12-21:
Add them here: https://drive.google.com/drive/folders/0B1zeRbBVLXhzZ0Q1TkxtbUxIcEU

I added one from my garage

jim-v, 2017-12-29:
If you are not sure which topics to record, `rosbag record -a` will capture all of them.

rohbotics, 2017-12-30:
I added a bag file from the 'ground truth' aruco test board I made to the google drive. It has various different lighting conditions. We should run a fiducial slam test on this bag and compare it to the design dimensions of the board. 

Here is the test board design:
[image]

The dimensions are from fiducial center line to center line. The center fiducial is angled 45 degrees and the other angled fiducial is at 30 degrees.

louislim70, 2018-01-01:
I wrote bellow in the closed issue.
I can detect above id of all markers .. 240, 241, 242, 243, 244.
Why can't i detect the markers which create by "http://terpconnect.umd.edu/~jwelsh12/enes100/markergen.html"? What is different?
================== bellow =============
When i tested, the aruco_detect package in fiducials could not detect markers created by "http://terpconnect.umd.edu/~jwelsh12/enes100/markergen.html".
But the marker of the test image(tag_01_d7_14cm.png) in aruco_detect/test/test_images was detected. The shape of two images that marker id is number 1 was very different.
The dictionary of two images is same as aruco::DICT_5X5_250.

Please tell me what is  different.

I also tested script (create_markers.py) to make some aruco markers.
I don't know if i want make the marker of aruco::DICT_5X5_250.
So, just typed like "create_marker.py 1 2 marker_pdf_file".
And then, there is a pdf file with 2 markers (8x8)which was not detected  in same directory.

What's problem??

jim-v, 2018-01-01:
@louislim70 Can you please create a new issue to ask your question:
https://github.com/UbiquityRobotics/fiducials/issues/new
I'm happy to help, but it is confusing to everyone if your question is mixed in with a conversation about a different topic.

rohbotics, 2018-01-21:
@jim-v Can you review the bags in the folder and see if that is 'good enough' to close this issue?

rohbotics, 2018-10-08:
Closing this issue, because the current ground truth board bag file is good enough for now. For anything more this issue becomes unclose-able.

closed 2018-10-08 by rohbotics