UbiquityRobotics/fiducials #89: Question about using multiple markers

nav-go, 2017-10-03:
Conceptually, how is the camera pose estimated from multiple markers? Does having multiple markers improve the pose estimate?

rohbotics, 2017-10-04:
Having multiple markers visible does improve the pose estimate.

Currently the default mode is to calculate the transform form the fiducial to the camera for each detected fiducial, and estimate an error of that measurement. These transforms are then combined, and are weighted based on the error in the measurement, as well as the estimated error of the fiducial in the map  to get the map to camera pose estimate.

@jim-v Can elaborate if you want.

nav-go, 2017-10-06:
I guess my question is : how are they combined if each marker is at a different position and orientation relative to the camera?

Another question : I've got the package running as per the instructions in the wiki and I am getting a pose estimate from /fiducial_pose with 1 marker placed at about half a meter from camera. 

I should be getting a pose estimate which says z=0.5 right? My pose estimate says 0.12. What is going wrong here? My camera is calibrated and I've set the fiducial_len of the marker as the length of the side of the black marker area.

jim-v, 2017-10-06:
@nav-go each of the fiducial markers is at a known 6 Degrees of Freedom pose in the map frame.  That pose corresponds to the transform from the map to that marker. For each detected marker, we get the pose of the camera in that marker's co-ordinate system.  By multiplying the transforms from map to marker and marker to camera, we get an estimate of the transform from the map frame to the camera (i.e. the camera's pose in the map frame) for each marker.  These estimates _should_ be consistent with each other.

For the second question, the topic `/fiducial_transforms` should contain the pose of the camera relative to the fiducial markers, which is where the 0.5 would be seen.  The topic `/fiducial_pose` is the robot's estimated pose in the map frame, which depends on the where the markers are in the map frame as well as the pose of the camera relative to them.  
 
If you would like us to look into this, could you please make a bag file with `rosbag record -a` and attach it to the issue.

nav-go, 2017-10-07:
Ah I think I get it it now. I'm trying to get the pose of the camera in an room. Please let me know if my approach is correct here:

I've posted a bunch of markers on the wall. I have to specify the location of one of the markers in the map.txt file. I plan to set it to be my origin in the map frame. So write 101 0 0 0 0 0 0 0 0  in the file.

To specify multiple markers, i can now say 102 1 0 0 0 0 0 0 0 to specify that the second marker is 1 meter to the right?

Also what is the fid 0 marker that is created?

While trying to look at it in Rviz, There are only transforms between the map and odom. The camera is related to the fiducials on a different tf tree. How do I get the camera to show up in Rviz?

Thanks for your help!

jim-v, 2017-10-07:
You could specify the pose of marker 102, but the software is intended to do Simultaneous Mapping and Localization, so it would build the map itself as it sees multiple markers.

The marked fid 0 is a combined estimate of the camera pose based upon all the markers.

In order to have the camera position show up, you need to specify the where the camera is on the robot, which you can do with `static-transform-publisher`.

nav-go, 2017-10-07:
I specified a transform between the camera and a base_link, but it is still not connected to odom. Is it supposed to be?

jim-v, 2017-10-07:
The default configuration is setup for use on a robot base, where the transform from odom to base_link is published.  fiducial_slam publishes the transform from map to odom, such that map to base_link is correct.  This requires looking up the transform from odom to base_link.  If you don't want that behavior, set the odom_frame to "".

jim-v, 2017-10-07:
The format of the map file is described in the documentation: http://wiki.ros.org/fiducial_slam

nav-go, 2017-10-08:
Thanks Jim.
Another Question regarding setting up markers and Rviz:

I've designated one corner of the room as the origin and setup the first marker 101 on a wall at a known location and orientation from origin. I can specify this in the map file. In rviz I can see the location change with rspect to map frame, but the orientation seems to not change properly. The markers keep facing down and changes the set pan/tilt angle by itself.  Is it hard coded somewhere that the markers are on ceiling? How do I change this?

jim-v, 2017-10-08:
Nothing in the code makes any assumptions about the pose of the markers - they are full 6DOF poses.
When you add an entry to the map manually, you define a co-ordinate system relative to the marker.  It may be that the orientation of that co-ordinate system is not what you intended.  You can start with an empty map (`rm ~/.ros/slam/map.txt`) and `fiducial_slam` will auto initialize the map by calculating a pose for the closest marker such that `base_link` is at the origin.  I recommend doing this once to see what the orientation of the marker is, then adjust the translation to put the origin in the corner of your room.

rohbotics, 2017-11-09:
@nav-go I am closing this issue since it seems your questions have been answered. Feel free to reopen or create a new issue if you have more questions.

Good luck with whatever project you are using `fiducials` for.

closed 2017-11-09 by rohbotics