UbiquityRobotics/fiducials #90: aruco_detect & fiducial_detect process segfaults with USB cameras

Hodapp87, 2017-10-09:
I have been trying to use USB cameras with the ArUco detection. What I am running into with both aruco_detect and fiducial_detect is that the process dies with exit code -11. With `aruco_detect` it dies immediately with:

```
[ INFO] [1507580447.119276852]: Aruco detection ready
[aruco_detect-1] process has died [pid 229, exit code -11, cmd /opt/ros/kinetic/lib/aruco_detect/aruco_detect /camera/compressed:=/usb_cam/ima
ge/compressed /camera_info:=/usb_cam/camera_info __name:=aruco_detect __log:=/root/.ros/log/4cf040da-ad2f-11e7-abe2-0242ac110002/aruco_detect-
1.log].
log file: /root/.ros/log/4cf040da-ad2f-11e7-abe2-0242ac110002/aruco_detect-1*.log
```

and with `fiducial_detect` it loops:

```
process[fiducial_detect-1]: started with pid [418]
[ INFO] [1507580566.238885018]: Fiducials Localization ready
camera intrinsics:
0.000000 0.000000 0.000000 
0.000000 0.000000 0.000000 
0.000000 0.000000 0.000000 

[fiducial_detect-1] process has died [pid 418, exit code -11, cmd /opt/ros/kinetic/lib/fiducial_detect/fiducial_detect /camera/compressed:=/us
b_cam/image/compressed /camera_info:=/usb_cam/camera_info __name:=fiducial_detect __log:=/root/.ros/log/4cf040da-ad2f-11e7-abe2-0242ac110002/f
iducial_detect-1.log].
log file: /root/.ros/log/4cf040da-ad2f-11e7-abe2-0242ac110002/fiducial_detect-1*.log
[fiducial_detect-1] restarting process
```
and I will then see in dmesg:

```
[459309.981281] aruco_detect[7187]: segfault at 0 ip 000000000041ac90 sp 00007ffc833be428 error 4 in aruco_detect[400000+4b000]
[459429.206708] fiducial_detect[7399]: segfault at 0 ip 00007faa401d1770 sp 00007ffd139c34b0 error 4 in librpp_pose.so[7faa401b2000+29000]
```

I reproduced this on a stock Ubuntu 17.04 configuration, as well as on Docker. I tried two different USB cameras using `usb_cam` (I observed the same results with `libuvc_camera`, however, I had neglected to write down details of my test). The two cameras are:

```
Bus 001 Device 011: ID 041e:4097 Creative Technology, Ltd Live! Cam Chat HD [VF0700]
Bus 001 Device 012: ID 0c45:6341 Microdia Defender G-Lens 2577 HD720p Camera
```

To reproduce, start from the Dockerfile below:

```docker
FROM ros:kinetic-ros-base
RUN apt-get update && apt-get install -y \
    tmux \
    ros-kinetic-usb-cam \
    ros-kinetic-fiducials \
    ros-kinetic-image-view \
    && rm -rf /var/lib/apt/lists/*
```

I used a script something like:

```
xhost local:root
docker run -ti --rm -e DISPLAY=$DISPLAY --device /dev/video0 -v /tmp/.X11-unix:/tmp/.X11-unix ros_kinetic_desktop
```

Then run the following in separate terminals (tmux?):

```
roslaunch usb_cam usb_cam-test.launch
roslaunch aruco_detect aruco_detect.launch camera:=/usb_cam
```

Or the same with `aruco_detect` replaced with `fiducial_detect`. The first command should open a window showing images streaming in; the second should trigger the errors.

jim-v, 2017-10-09:
It looks like your camera is not calibrated.  See http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration for details of how to do this.

Clearly, `aruco_detect` and `fiducial_detect` should handle this condition more gracefully than with a segmentation fault.

rohbotics, 2017-10-10:
@Hodapp87 Thanks for such a detailed issue report, and a Docker reproduction of the issue.

@jim-v I made a PR #91, that should fix the crashing symptom.

rohbotics, 2017-10-10:
@Hodapp87 PR #91 was merged, so the code should at least not segfault with no camera calibration in the next release.

Let us know if you were able to calibrate your camera and use aruco_detect.

Hodapp87, 2017-10-10:
Oh, interesting. I wasn't aware calibration was a requirement; I had just figured that uncalibrated cameras would give nonsense results, but still at least detect ArUco markers in the frame.

Running with the following seems to be working (my test case had a small mistake):

```
roslaunch aruco_detect aruco_detect.launch camera:=/usb_cam image:=image_raw
```

jim-v, 2017-10-10:
Calibration is only a requirement if you want to do 3D pose estimation of the markers.  If you just wacnt the image co-ordinates, then you can set the `do_pose_estimation` parameter in `aruco_detect` or the `estimate_pose` parameter in `fiducial_detect` to false.  Sorry for the inconsistency in parameter names!

rohbotics, 2017-10-12:
@Hodapp87 As it seems your issue is resolved, I am closing the issue.

If you have anymore questions feel free to reopen, or create a new issue. Good luck with whatever you are using `aruco_detect` for.

closed 2017-10-12 by rohbotics