UbiquityRobotics/fiducials #95: Using multiple cameras

nav-go, 2018-01-06:
Is it possible to localize multiple cameras at the same time within a single map or even multiple maps? This feature would be very useful. 

I'm basically trying to run two instances of fiducials in parallel with two cameras by putting them in different namespaces but since aruco_detect publishes fiducial_images,vertices and transforms to a single topic, this is a problem.

Is there a recommended way to do this using ROS remapping or will this require changing code within fiducials?

jim-v, 2018-01-06:
To use multiple cameras with a single map would require a change to the map class of `fiducial_slam` so that the transform from `base_link` to the camera is looked up using the `frame_id` field of the transform message.  Then, the estimates would be combined at `base_link`, rather than at the camera, as they are now.  It's not a big change, but it's not currently in our plans.

If you wanted to use a separate map for each camera, that could be done by renaming the topics and running an instance of `fiducial_slam` for each camera.  It's not clear what the benefit of that arrangement would be.

nav-go, 2018-01-10:
Thanks, I was able to run two fiducial nodes in parallel by remapping the topics.

closed 2018-01-10 by nav-go

dj4park, 2019-08-16:
Could someone elaborate on the steps that I need to take for me to detect a single aruco marker from two different cameras? 

Currently, we are placing a marker on a robot and localizing it through the cameras.