UbiquityRobotics/fiducials #98: Question about fiducial pose

nav-go, 2018-01-24:
While testing this package, I noticed that the transforms computed by the aruco_detect module is much more stable and less noisy than the final pose computed by the fiducial_slam module. In particular, the left-right and up-down directions (considering moving towards the marker to be forward) are much noisier than forward/backward.

The noise seems to be introduced after the transforms pass through the fiducial_slam package. Is this expected behavior?

jim-v, 2018-01-25:
There may be some jitter in the final pose computed by fiducial_slam. If the camera is not well calibrated, it's possible that the pose estimates from different fiducials are not consistent with each other.  Can you record a bag file of all related topics (or `rosbag record -a` if you are not sure which), so I can take a look.  You can upload it here: https://drive.google.com/drive/folders/0B1zeRbBVLXhzZ0Q1TkxtbUxIcEU

jim-v, 2018-01-25:
Information about calibrating a camera: http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

closed 2018-07-22 by rohbotics