UbiquityRobotics/fiducials #104: compilation crash on cv::String::deallocate

GuillaumeHauss, 2018-04-17:
Hi all, 
I tried to compile your package from source using catkin on ROS Kinetic and Ubuntu 16.04. 
However the compilation crashed on your package. Here is the log trace:

```
CMakeFiles/create_marker.dir/src/create_marker.cpp.o: In function `cv::String::String(char const*)':
create_marker.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x3f): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/create_marker.dir/src/create_marker.cpp.o: In function `main':
create_marker.cpp:(.text.startup+0x4c): undefined reference to `cv::CommandLineParser::CommandLineParser(int, char const* const*, cv::String const&)'
create_marker.cpp:(.text.startup+0x59): undefined reference to `cv::String::deallocate()'
create_marker.cpp:(.text.startup+0x77): undefined reference to `cv::CommandLineParser::about(cv::String const&)'
create_marker.cpp:(.text.startup+0x81): undefined reference to `cv::String::deallocate()'
...
```
The fix I found was the following:
Change `find_package(OpenCV REQUIRED)` to `find_package(OpenCV 3 REQUIRED)`
and `${OpenCV_LIB}` to `${OpenCV_LIBRARIES}` every time it was used

I guess there are additionnal checks to perform in the CMakeList.txt but as I'm not an expert I don't want to draft something wrong

jim-v, 2018-04-17:
Thanks for reporting this.  We will take a look at it.  Note that it does build for others on ROS Kinetic and Ubuntu 16.04, so we need to make sure we don't break that.

rohbotics, 2018-07-27:
Closing due to age. 

Hope you had some good luck getting this working. Feel free to re-open if the bug still exists for you.

closed 2018-07-27 by rohbotics