UbiquityRobotics/fiducials #107: Question regarding global map optimization.

fabianbl, 2018-05-31:
Very useful and lightweight library, thank you.

One thing that I'm not sure if I got it right from the code is how the map update works: It somehow seems that the poses of the fiducials observed both in the previous and the current frame are updated taking into account the transform variances, is this correct? But there is no global optimization (e.g. non-linear optimization over all corner reprojection errors for all fiducials), right?

jim-v, 2018-05-31:
That's correct - there is currently no global optimization.  It's something we have discussed in #62.  If you would like to continue this discussion, please could you do it in that issue.

closed 2018-05-31 by jim-v

rohbotics, 2018-05-31:
Saving enough data to do a global optimization such as bundle adjustment is something we hope to implement in the future. 

Feel free to ask anymore questions you might have, good luck with whatever you are building using fiducials.

Also PRs are welcome :)

fabianbl, 2018-05-31:
@jim-v @rohbotics Thanks for your feedback :-)