UbiquityRobotics/fiducials #116: Offset on the fiducial transforms

elisabethwelburn12, 2018-09-21:
Hi all,

I am trying to use this program to perform SLAM on my virtual quadcopter on gazebo. I am getting a weird slight offset for all the fiducial transforms produced. I have calibrated the gazebo camera and it has not improved. There are screenshots for both the gazebo simulation and the rviz visualisation. The fid0 transform is off centre and the robot is slightly falling through the floor and I am unsure of how to fix this.
[image]
[image]

The launch files for both the aruco detect and fiducial_slam nodes are as follows:
```
<node pkg="tf" type="static_transform_publisher" name="base_footprint_to_optical_cam_frame"  args="-0.4 0.0 0 1.57 0 -1.57 /base_link /cam_optical_frame 100"/> 
<arg name="camera" default="/cam/camera"/>
  <arg name="image" default="image"/>
  <arg name="transport" default="compressed"/>
  <arg name="fiducial_len" default="0.44237"/> 
  <arg name="dictionary" default="7"/>
  <arg name="do_pose_estimation" default="true"/>

  <node pkg="aruco_detect" name="aruco_detect"
    type="aruco_detect"  respawn="false">
    <param name="image_transport" value="$(arg transport)"/>
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="$(arg fiducial_len)"/>
    <param name="dictionary" value="$(arg dictionary)"/>
    <param name="do_pose_estimation" value="$(arg do_pose_estimation)"/>
    <remap from="/camera/compressed"
        to="$(arg camera)/$(arg image)/$(arg transport)"/>
    <remap from="/camera_info" to="$(arg camera)/camera_info"/>
 
  </node>

  <arg name="map_frame" default="world"/> 
  <arg name="base_frame" default="base_link"/>
  <arg name="publish_tf" default="false"/>
  <arg name="future_date_transforms" default="0.0"/>
  <arg name="publish_6dof_pose" default="true"/>

  <node type="fiducial_slam" pkg="fiducial_slam" 
    name="fiducial_slam" output="screen">

    <param name="map_file" value="$(env HOME)/.ros/slam/map.txt" />
    <param name="map_frame" value="$(arg map_frame)" />
    <param name="odom_frame" value="" /> 
    <param name="base_frame" value="$(arg base_frame)" />
    <param name="future_date_transforms" value="$(arg future_date_transforms)" />
    <param name="publish_6dof_pose" value="$(arg publish_6dof_pose)" />
    <param name="do_pose_estimation" value="$(arg do_pose_estimation)"/>
    
    <param name="fiducial_len" value="$(arg fiducial_len)"/>
    <remap from="/camera_info" to="$(arg camera)/camera_info"/>
    <remap from="camera_frame" to="cam_optical_frame"/>

  </node>
```
I don't know if this is just a bug or just a transformation error, Thanks for help in advance.

jim-v, 2018-09-21:
One thing to note is that when the map is auto-initialized, the pose of the fiducial closest to the robot is calculated such that the robot is at (0, 0, 0).   Is the map being initialized with the quad-copter in flight, or on the ground?   Note that you also have the option of manually initializing the map by creating a single entry in `map.txt` as described in http://wiki.ros.org/fiducial_slam?distro=kinetic

elisabethwelburn12, 2018-09-21:
I manually initialise the map text file and input all the fiducials within it and the markers then appear at the correct location relative to the world frame in RViz.  I start the nodes with the robot on the ground. I have tried initialising the map file with only one fiducial and two fidcucials and the offset remains.  I have noticed the Fid0 is offset to the world consistently and I wonder if this is impacting the fiducial transforms being incorrectly placed. It was said on this forum that the Fid0 is the camera frame relative to the markers and yet it maps the cam_optical_frame somewhere else? Thanks for the fast reply I really appreciate it !

elisabethwelburn12, 2018-09-26:
Okay, I think I know the cause of this issue, it is because the cam_optical_frame is being estimated at floor height by the fiducial markers. This causes the robot to tilt at an angle, due to the static state publisher publishing a fixed distance between the base link and the camera frame. I am unsure of how to fix this, I have added a z offset on the static transform publisher as above and it just causes the robot to fall through the floor within Rviz. Is there a way to adjust the height of the estimated camera frame? Thanks

jim-v, 2018-09-26:
I'm not sure what you are asking - cam_optical_frame does not appear to be at floor height in the image above.  The pose of the camera is estimated from the position of the fiducials in the world (from the map), transformed by the fiducial to camera transform from the pose estimation of the fiducials.  The final estimate of the robot's position is constrained to be on the floor if publish_6dof_pose is not set, but it is in the launch file above.

elisabethwelburn12, 2018-09-26:
There is a distortion of where cam_optical_frame is spawned within RViz in comparision to the gazebo model. These screenshots below show this a lot clearer.

[image]
[image]

vishalsleotechsa, 2019-07-23:
hi all, 
got Oscillation in robot pose when using fiducial slam.
I have to use the fiducial slam for robot localization using fiducial markers, I have to download the fiducial package from this link https://github.com/UbiquityRobotics/fiducials.git, we use the ROS distribution which is indigo and Kinect camera. And the markers on the wall and the camera is front-facing like human structure. When we start the fiducial_slam node for the localization of robot, we get the small oscillation in the robot pose estimation, when moving the robot or increase the distance between the marker and camera the localization of robot oscillation is increases more like 1 meter. so what is the root cause of this issue?