UbiquityRobotics/fiducials #121: Localization mode

rdaems, 2018-10-02:
Hi

Is it possible to work in localization mode? The position of the markers keeps changing on the map because accuracy is low and I would like to keep them fixed, so I can be sure other stuff I define in the map frame stays fixed to the markers.

Regards

Rembert

jim-v, 2018-10-02:
There are two new parameters in the source for `fiducial_slam` on github:

`read_only_map` if set to true, the map will not be modified (I think this is what you mean by localization mode).

`sum_in_quadrature` if set to true, a different algorithm is used for combining pose estimates, that may be more stable in some circumstances.

They will make it into a release in due course, but if you want to try them, the quickest way is to get the source.

rdaems, 2018-10-02:
Hi @jim-v 

Now no new markers are added to the map, that's good. But the markers still change relatively to the map frame and to each other. Is this meant to happen?

I would like to construct a map once, and then use it without changing it.

This is my launch file:

```
<launch>
    <arg name="dictionary"          value="3"/>
    <arg name="fiducial_len"        value="0.140"/>

    <node type="rviz" name="rviz" pkg="rviz" args="-d $(find dribble_slam)/rviz/fiducial_slam.rviz" />

    <node pkg="aruco_detect" name="aruco_detect_robot" type="aruco_detect" respawn="false" output="log" launch-prefix="">
        <param name="image_transport" value="compressed"/>
        <param name="publish_images" value="true" />
        <param name="fiducial_len" value="$(arg fiducial_len)"/>
        <param name="dictionary" value="$(arg dictionary)"/>
        <param name="do_pose_estimation" value="false"/>
        <param name="adaptiveThreshWinSizeMin" value="3"/>
        <param name="adaptiveThreshWinSizeMax" value="23"/>
        <param name="adaptiveThreshWinSizeStep" value="10"/>

        <remap from="/camera/compressed" to="/camera/color/image_raw/compressed"/>
        <remap from="/camera_info" to="/camera/color/camera_info"/>
        <remap from="/fiducial_images" to="/fiducial_images"/>
    </node>

    <node type="fiducial_slam" pkg="fiducial_slam" output="screen" name="fiducial_slam">
        <param name="map_file" value="$(env HOME)/.ros/slam/map.txt" />
        <param name="map_frame" value="map" />
        <param name="odom_frame" value="" />
        <param name="base_frame" value="camera_link" />
        <param name="future_date_transforms" value="0.0" />
        <param name="publish_6dof_pose" value="false" />
        <param name="do_pose_estimation" value="true"/>
        <param name="fiducial_len" value="$(arg fiducial_len)"/>

        <param name="read_only_map" value="true"/>
        <param name="sum_in_quadrature" value="true"/>

        <remap from="/camera_info" to="/camera/color/camera_info"/>
    </node>
</launch>
```

rohbotics, 2018-10-08:
You you running the latest code from github or are you using the version from apt?

rdaems, 2018-10-08:
I'm running latest code on github

jim-v, 2018-10-08:
In rviz, you should see red or green cuboids for the positions of the fiducials in the map - these should not move.  There will also be a set of axes shown with a name like fidxxx for each observation of a fiducial marker.   Can you confirm if the cuboids are moving (they shouldn't).   Thanks!

rohbotics, 2018-10-08:
Are the markers that you are talking about the tfs or the cubiods in rviz? The tfs are published by aruco_detect and are just the position estimates of the fiducial relative to the camera.

rdaems, 2018-10-11:
Hi

Yes they were the tfs, I thought they were published by the slam.

Thanks for the explanation, I'll close this now

closed 2018-10-11 by rdaems