UbiquityRobotics/fiducials #136: How to use Aurco_detect  without raspicam?

chrissunny94, 2018-10-24:
My launch file looks like the below .

`<launch>
  <arg name="mapping_mode" default="false"/>

 
  
  
  <include file="$(find aruco_detect)/launch/aruco_detect.launch">
    <arg name="camera" value="/camera/rgb"/>
    <arg name="image" value="image_raw"/>
    <arg name="transport" value="raw"/>
    
    
    <arg name="fiducial_len" value="0.140" />
  </include>

  
  <include file="$(find fiducial_slam)/launch/fiducial_slam.launch">
    <arg name="map_frame" value="map"/>
    <arg name="odom_frame" value="odom"/>
    <arg name="base_frame" value="base_link"/>
    <arg name="future_date_transforms" value="0.5"/>
  </include>
</launch>`

The topics concerned with the camera (Currently a Kinect is publishing the below topics )

`/camera
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/points
/camera/parameter_descriptions
/camera/parameter_updates
/camera/rgb/camera_info
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/imagezero
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
`

chrissunny94, 2018-10-24:
`
<launch>
  <arg name="mapping_mode" default="false"/>

 
  
  
  <include file="$(find aruco_detect)/launch/aruco_detect.launch">
    <arg name="camera" value="/camera/rgb"/>
    <arg name="image" value="image_raw"/>
    <arg name="transport" value="raw"/>
    
    
    <arg name="fiducial_len" value="0.140" />
  </include>

  
  <include file="$(find fiducial_slam)/launch/fiducial_slam.launch">
    <arg name="map_frame" value="map"/>
    <arg name="odom_frame" value="odom"/>
    <arg name="base_frame" value="base_link"/>
    <arg name="future_date_transforms" value="0.5"/>
  </include>
</launch>`

rohbotics, 2018-10-24:
Hey @chrissunny94 

Something like this should work
```
<arg name="camera" value="/camera/rgb"/>
<arg name="image" value="image_raw"/>
<arg name="transport" value="raw"/>
```

chrissunny94, 2018-10-24:
@rohbotics , Thanks man , solved it .