UbiquityRobotics/fiducials #137: aruco_detect does not detect all visible marker

nihad-s, 2018-10-26:
Its very strange behavior where the same marker (on the wooden pole) gets detected from quite far away but once I get close and the marker/camera are still it is not getting detected even though all the marker are clearly better visible.

[image]

Putting a white background (white paper) behind the marker or moving it makes it again detectable. But I am not sure what the specific cause is so I can fix it or avoid it. Which specific setup would make it most detectable, painting the pole white, increasing white boarder of the marker, changing the detector parameters?

The launch file I am using right now is:

`  <arg name="camera" default="camera/color"/>
  <node pkg="aruco_detect" name="aruco_detect" type="aruco_detect" output="screen" respawn="false">
    <param name="image_transport" value="raw"/>
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="0.19"/>
    <param name="dictionary" value="3"/>
    <param name="do_pose_estimation" value="true"/>
    <remap from="/camera" to="$(arg camera)/image_raw"/>
    <remap from="/camera_info" to="$(arg camera)/camera_info"/>
  </node>
`

I have also bag of the image_raw and camera_info.
https://files.fm/u/cbscbrrn

jim-v, 2018-10-28:
The detector is looking for a wider white border around the image than you have in the undetected fiducial.  It is possible to adjust the parameters.   They can be adjusted with [dynamic reconfigure] (http://wiki.ros.org/rqt_reconfigure) so you can find parameters that work for you and then set in the configuration or launch file.

rohbotics, 2018-10-28:
The minimum white border we recommend is slightly larger than 1 square of the marker. 

The lines in the file from `create_markers.py` represent the recommended minimum border.

closed 2018-11-03 by rohbotics