UbiquityRobotics/fiducials #138: Split detection and pose estimation in aruco_detect

rohbotics, 2018-10-28:
Have aruco_detect only detect the markers in the image_callback and publish the vertices. 

Then have it subscribe to its own vertices publish and do estimation in that callback.

This should create a better code split, and allow us to write tests of the pose estimation. Also then we can try using an `AsyncSpinner` on multiple threads to see if there is a performance improvement.

jim-v, 2018-10-28:
Having tests of the pose estimation seems good.

I don't think there will be much of a performance improvement from this refactor, since the expensive OpenCV routines use multiple threads.

rohbotics, 2018-11-04:
The problem I see with this now is the published images. These require both the vectors and the axis, so the image needs to be copied around.

closed 2020-12-15 by tp4348