UbiquityRobotics/fiducials #139: Make sure cariosvg is available

jim-v, 2018-11-10:
```
$ rosrun aruco_detect create_markers.py 
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/aruco_detect/create_markers.py", line 8, in <module>
    import cairosvg
ImportError: No module named cairosvg
```
This could be addressed by calling `checkCmd` or with package dependencies.

rohbotics, 2018-11-10:
This is already in there:    
```checkCmd("cairosvg-py2", "python-cairosvg")```

now that this PR is in: https://github.com/ros/rosdistro/pull/19245/files we should just add it as a rosdep, along with joblib https://github.com/ros/rosdistro/pull/19244

jim-v, 2018-11-10:
It shoukd have the import statement for cairo in a try block, otherwise the
script exits before the checkCmd if cairo is not installed.

closed 2018-11-12 by rohbotics

juanea, 2019-01-31:
I still have this problem.

$ rosrun aruco_detect create_markers.py 
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/aruco_detect/create_markers.py", line 8, in <module>
    import cairosvg
ImportError: No module named cairosvg

What should I do?

Thanks

anfederman, 2019-04-12:
I just ran into this now.  If you try to run by changing to python3, the script fails. So no markers for me

jim-v, 2019-04-12:
The issue you ran into is #161.

PR #140 (2018-11-12): Added rosdeps for cairo and joblib
Fixes #139