UbiquityRobotics/fiducials #144: Aruco_detect detects 1 marker before dying

LapinCodeur, 2019-01-03:
I have a small issue with Aruco_detect, when I launched it, I got this :

`projet-d@projetd-VirtualBox:~/ROSCube$ roslaunch aruco_detect aruco_detectTest.launch
... logging to /home/projet-d/.ros/log/c9d5a85c-0f6b-11e9-bf06-0800279b5cc6/roslaunch-projetd-VirtualBox-4796.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://projetd-VirtualBox:33619/

SUMMARY
========

PARAMETERS
 * /aruco_detect/dictionary: 16
 * /aruco_detect/do_pose_estimation: True
 * /aruco_detect/fiducial_len: 0.14
 * /aruco_detect/fiducial_len_override: 
 * /aruco_detect/ignore_fiducials: 
 * /aruco_detect/image_transport: raw
 * /aruco_detect/publish_images: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    aruco_detect (aruco_detect/aruco_detect)

ROS_MASTER_URI=http://localhost:11311

process[aruco_detect-1]: started with pid [4813]
[ INFO] [1546529795.658842071, 610.994000000]: Aruco detection ready
[ INFO] [1546529795.997013842, 611.276000000]: Got image 5
[ INFO] [1546529796.154800864, 611.434000000]: Detected 1 markers
[aruco_detect-1] process has died [pid 4813, exit code -11, cmd /home/projet-d/ROSCube/devel/lib/aruco_detect/aruco_detect /camera_info:=/realsense/camera/color/camera_info /camera:=/realsense/camera/color/image_raw __name:=aruco_detect __log:=/home/projet-d/.ros/log/c9d5a85c-0f6b-11e9-bf06-0800279b5cc6/aruco_detect-1.log].
log file: /home/projet-d/.ros/log/c9d5a85c-0f6b-11e9-bf06-0800279b5cc6/aruco_detect-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
The log file is empty. I am not quite sure about my launch file, I have tried to modify it to run with the published topic of my Realsense camera that come from my simulation. The topic from the camera works fine.
My launch file : 
`
<launch>
  
  <arg name="camera" default="/realsense/camera/color"/>
  <arg name="image" default="image"/>
  <arg name="transport" default="raw"/>
  <arg name="fiducial_len" default="0.14"/>
  <arg name="dictionary" default="16"/>
  <arg name="do_pose_estimation" default="true"/>
  <arg name="ignore_fiducials" default="" />
  <arg name="fiducial_len_override" default="" />

  <node pkg="aruco_detect" name="aruco_detect" type="aruco_detect" output="screen" respawn="false">
    <param name="image_transport" value="$(arg transport)"/>
    <param name="publish_images" value="true" />
    <param name="fiducial_len" value="$(arg fiducial_len)"/>
    <param name="dictionary" value="$(arg dictionary)"/>
    <param name="do_pose_estimation" value="$(arg do_pose_estimation)"/>
    <param name="ignore_fiducials" value="$(arg ignore_fiducials)"/>
    <param name="fiducial_len_override" value="$(arg fiducial_len_override)"/>
    <remap from="/camera_info" to="$(arg camera)/camera_info"/>
    <remap from="/camera" to="$(arg camera)/image_raw"/>
  </node>
</launch>`

LapinCodeur, 2019-01-04:
I have two problems, I can't change do_pose_estimation to false, if I do it in the launch file, it is still appearing as true in parameters when I run the launch file. And the error that is breaking down my node is due to this function `void FiducialsNode::imageCallback(const sensor_msgs::ImageConstPtr & msg)`
Inside of this :
```
if (!haveCamInfo) {
        if (frameNum > 5) {
            ROS_ERROR("No camera intrinsics");
        }
        return;
}
```
It seems that frameNum is below 5.
So, I just got the return and crashing.

LapinCodeur, 2019-01-07:
I think that the error is due to the topic camera_info that is incomplete in the simulation of the realSense camera. I have try with a more simple camera and everything was fine.
```
header: 
  seq: 813
  stamp: 
    secs: 380
    nsecs: 178000000
  frame_id: "color"
height: 480
width: 640
distortion_model: ''
D: []
K: [554.3826904296875, 0.0, 320.0, 0.0, 554.3826904296875, 240.0, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [554.3826904296875, 0.0, 320.0, 0.0, 0.0, 554.3826904296875, 240.0, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

rohbotics, 2019-02-10:
Since the solution seems to be just calibrating the camera, i am closing this issue.

jim-v, 2019-02-10:
I think that the case where there is calibration data, but it is invalid should be handled more gracefully.  I will implement a fix, in the next week or two.

rohbotics, 2019-02-10:
That seams reasonable

torydebra, 2022-01-17:
Hi, I do not know if this is still active, but I faced a similar issue with a (simulated) d435 from https://github.com/pal-robotics/realsense_gazebo_plugin

The issue is here:
https://github.com/UbiquityRobotics/fiducials/blob/1ae8a13d56fafa436aca181da451c1bf808a288c/aruco_detect/src/aruco_detect.cpp#L320-L322

the node crashed because the simulated camera does not fill the D with anything.

For now I solved modifying the code of the simulated camera, filling the field with zeros as aruco_detect expect