UbiquityRobotics/fiducials #145: Fiducial marker visual lagging and image jumping isuue

aaryansagar, 2019-01-28:
Hi,
While tracking the markers and publishing the image, the image is jumping and it only shows when the marker is kept static. While moving the marker quickly, the continuous movement image of the marker is not being published and it appears as the image is jumping. However, If I move it slowly then it can show a continuous image. Also, It is happening in x and y-direction only and its okay in z-direction. Can you please help me solve this issue? What do you think might be the issue?

aaryansagar, 2019-01-28:
Thank you for your reply. So, in my case, I want to use it where the camera is fixed but markers are attached to the object and the object is moving. I guess thats the problem as according to you the fiducial package is designed where marker is static and camera is moving. If thats the case can you suggest which other package can be used in my case or if there is any other way to fix this issue?

Also, x, y and z direction was with respect to the coordinate system of the marker where left/right, up/down is x and y axis respectively and towards and away from camera is z-axix. 
Thank you!!

jim-v, 2019-01-28:
There are two nodes involved: `aruco_detect` which detects the markers and publishes their pose relative to the camera, and `fiducial_slam` which deals with the map.  In your case, you don't want `fiducial_slam`, but maybe `aruco_detect` is still useful.

Note that in the coordinate system of the marker, z is normal to the plane of the camera and is not related to the distance from the camera.

aaryansagar, 2019-01-28:
Actually, I am only using aruco_detect in my program. Its showing the lag in the RVIZ section. Do you think it might have to do something with RVIZ code because of which its causing the lag for movement of marker at certain speed?

rohbotics, 2019-01-28:
The lag could be from the processing, what kind of system are you running this on?
Jim, could you suggest the parameters to tune to reduce CPU requirements?

Also if you are trying to track fast moving fiducials, make sure that your camera has a low shutter speed so that you don't have too much motion blur. What camera are you using?

aaryansagar, 2019-01-28:
I am using Ubuntu 16.04 on VMWare and have allowed 16GB of RAM and 4 core processors to the VMWare. 
[image]

Also, I have tried it with both webcam and IDS UI3160 camera with max 255fps.

jim-v, 2019-01-28:
You can get a measure of the speed with `rostopic hz /fiducial_transforms`   The processing can be sped up in various ways, all with some consequences.   One way is to work with a smaller image, if that is acceptable, other ways involve changing the detection parameters.   Let's start by determining what speed you are getting and what is desired.

rohbotics, 2019-03-25:
@aaryansagar I am closing this because of lack of response.

Hope that you found a solution for your project. Feel free to reopen if desired.

closed 2019-03-25 by rohbotics