UbiquityRobotics/fiducials #148: publish_tf doesn't seem to be used in the example fiducial_slam.launch

jack-digilabs, 2019-02-06:
I see there is a boolean rosparam called "publish_tf" which I'd assume turns off the map->odom publishing however it doesn't look like it's used at all. Is this what it was intended for but didn't get implemented?

I didn't see that rosparam until after I already wrote some of my own code to that checks if tf_publish_interval is negative, then doesn't publish tf and it's working fine, but I'd be happy to switch it to use the rosparam and make a PR addressing that.

elisabethwelburn, 2019-02-18:
Hi, I'm currently trying to work on the same problem (ie publishing pose but not the tfs themselves). Can I ask, what did you change exactly? Did you just make the tf_publish_interval  negative? Could you attach this code? You'd be doing me a huge favour and also I'm happy to help implementing it :)

jack-digilabs, 2019-02-18:
Sure can! I won't be able to get to it immediately today but I'll be
submitting a PR before the end of the day (CST).

rohbotics, 2019-03-25:
Closed by #154

closed 2019-03-25 by rohbotics