UbiquityRobotics/fiducials #149: Build error with OpenCV 3.2

Tuebel, 2019-02-07:
I think it is the same build issue that was mentioned in a comment of #94.
The problem is that the missing members (see the log) were introduced in OpenCV 3.3.
In aruco_detect.cpp the macro `#if OPENCV_MINOR_VERSION==2` tries to prevent the error but the version.hpp from OpenCV only defines `#define CV_MINOR_VERSION 2`.
Changing the lines in aruco_detect.cpp fixes the issue for me.

Here is the cmake error:
```bash
Errors     << aruco_detect:make /home/tim/code_repo/realsense_ws/logs/aruco_detect/build.make.002.log                                                                              
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp: In member function ‘void FiducialsNode::configCallback(aruco_detect::DetectorParamsConfig&, uint32_t)’:
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:252:26: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_SUBPIX;
                          ^~~~~~~~~~~~~~~~~~~~~~
                          cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:252:58: error: ‘CORNER_REFINE_SUBPIX’ is not a member of ‘cv::aruco’
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_SUBPIX;
                                                          ^~~~~~~~~~~~~~~~~~~~
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:255:26: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_CONTOUR;
                          ^~~~~~~~~~~~~~~~~~~~~~
                          cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:255:58: error: ‘CORNER_REFINE_CONTOUR’ is not a member of ‘cv::aruco’
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_CONTOUR;
                                                          ^~~~~~~~~~~~~~~~~~~~~
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:259:24: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
        detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_NONE;
                        ^~~~~~~~~~~~~~~~~~~~~~
                        cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:259:56: error: ‘CORNER_REFINE_NONE’ is not a member of ‘cv::aruco’
        detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_NONE;
                                                        ^~~~~~~~~~~~~~~~~~
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp: In constructor ‘FiducialsNode::FiducialsNode(ros::NodeHandle&)’:
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:575:26: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_SUBPIX;
                          ^~~~~~~~~~~~~~~~~~~~~~
                          cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:575:58: error: ‘CORNER_REFINE_SUBPIX’ is not a member of ‘cv::aruco’
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_SUBPIX;
                                                          ^~~~~~~~~~~~~~~~~~~~
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:578:26: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_CONTOUR;
                          ^~~~~~~~~~~~~~~~~~~~~~
                          cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:578:58: error: ‘CORNER_REFINE_CONTOUR’ is not a member of ‘cv::aruco’
          detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_CONTOUR;
                                                          ^~~~~~~~~~~~~~~~~~~~~
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:582:24: error: ‘struct cv::aruco::DetectorParameters’ has no member named ‘cornerRefinementMethod’; did you mean ‘cornerRefinementWinSize’?
        detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_NONE;
                        ^~~~~~~~~~~~~~~~~~~~~~
                        cornerRefinementWinSize
/home/tim/code_repo/realsense_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:582:56: error: ‘CORNER_REFINE_NONE’ is not a member of ‘cv::aruco’
        detectorParams->cornerRefinementMethod = aruco::CORNER_REFINE_NONE;
                                                        ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/aruco_detect.dir/src/aruco_detect.cpp.o] Error 1
make[1]: *** [CMakeFiles/aruco_detect.dir/all] Error 2
make: *** [all] Error 2
```

closed 2019-02-08 by Tuebel

rajk97, 2019-12-13:
I'm a beginner and having the same issue. Could you please tell me how do you change the aruco_detect.cpp  file? Thanks!

Tuebel, 2019-12-13:
Simply replace `#if OPENCV_MINOR_VERSION==2` with `#define CV_MINOR_VERSION 2`.
But since this code is merged, this bug should not occur if you pull the latest kinetic-devel source.

rohbotics, 2019-12-13:
@rajk97 How are you trying to build the code? Where did you pull it from.

If you are running from the latest branch, you shouldn't be having this problem, as the fix was merged in.