UbiquityRobotics/fiducials #152: Increasing detection robustness using OpenCv default parameters

Tuebel, 2019-02-12:
Is there any reason why you are not using the OpenCv default DetectorParameters from this [page](https://docs.opencv.org/3.1.0/d1/dcd/structcv_1_1aruco_1_1DetectorParameters.html)?
I found the detection to be more reliable using those parameters with dynamic_reconfigure.

rohbotics, 2019-02-12:
We tuned the detector params for a balance of performance and robustness.

We should probably re-evaluate how much faster our detector params are.

Tuebel, 2019-02-13:
That makes sense to me.

I did not measure the performance but the node is still able to run at 30fps on a 1280x720 resolution.
So for me the performance is just fine.

rohbotics, 2019-02-13:
What are your running on? We were doing performance tuning for a raspberry pi, so you probably have much more computational horsepower.

Tuebel, 2019-02-13:
I'm running this on my work notebook, which has a 7th Gen i7 processor.
So in contrast to your raspberry pi setup I will probably never see a difference.

rohbotics, 2019-02-17:
Closing this now, thanks for reaching out

closed 2019-02-17 by rohbotics