UbiquityRobotics/fiducials #155: melodic release

mistoll, 2019-02-22:
What are your plans for releasing to melodic?

With https://github.com/UbiquityRobotics/fiducials/commit/7f9d4d7107e3432f24761c3500f342e2c748ecef a major issue on melodic is solved.

rohbotics, 2019-02-23:
@jim-v 
Do you think there are any big changes/cleanup things that we need to do? Doing them only for the melodic release may make sense

jim-v, 2019-02-23:
It builds without warnings and the tests pass under melodic, so I think it will be fine.

rohbotics, 2019-02-24:
Ok I will run the melodic release next week

rohbotics, 2019-03-29:
Hi there, sorry for the delay. I ran into an issue with how we define the opencv dependency and how we should define it for melodic.

The `opencv3` package no longer exists, so we need to change it to `libopencv-dev` for melodic onward. I wonder if there is a way to do this without a branch.

mistoll, 2019-03-29:
There's some documentation about jade to kinetic migration in http://wiki.ros.org/opencv3 (Dual Compatibility). They propose to depend on cv_bridge or image_geometry. IMO this should also work for kinetic to melodic migration. As fiducials already depends on cv_bridge you can simply remove the opencv3 dependency.

closed 2019-06-15 by rohbotics