UbiquityRobotics/fiducials #161: CairoSVG only with Python 3

rfeistenauer, 2019-04-01:
When using the command to create markers:
rosrun aruco_detect create_markers.py 100 112 fiducials.pdf

i get: ImportError: No module named cairosvg

Trying to install it results in another error:
CairoSVG does not support Python 2.x anymore. Please use Python 3 or install an older version of CairoSVG.

I already used marker creation some month ago, so I gues thats a new problem.

Even switching to older versions just causes new Errors, so maybe this dependency can be switched to something else?

jim-v, 2019-04-04:
We will look into an alternative.  It's not clear what the best solution is, since the OpenCV Python bindings are not available for Python3 with the standard ROS packages on Kinetic.

rohbotics, 2019-04-04:
What version of ubuntu are you on where you see this error?

rfeistenauer, 2019-04-04:
Ubuntu 16.04 with ROS kinetic

rohbotics, 2019-04-07:
I think svglib is a good solution, it will not be as fast as CairoSVG, but nowhere near as slow as Inkscape was with its long startup times.

I'll look into it

PR #165 (2019-04-13): Use subprocess for cairosvg to avoid lack of python2 support
Fixes #161

The slowdown seems to be 2x, because of the process creation overhead. 

@rfeistenauer You may want to try this out.
closed 2019-04-13 by rohbotics