UbiquityRobotics/fiducials #164: Get the plane coordinates of the fiducial markers

innovatim, 2019-04-04:
In my test environment, the fiducial markers are attached to the ceiling, so the fiducial markers are all on the same plane. I want to get the absolute coordinates of the plane of each marker (the first marker detected is set to the origin, the coordinates of other markers are relative to the coordinates of the origin). Is there a corresponding method?

jim-v, 2019-04-04:
The 6-DOF pose of each marker is stored in the file ~/.ros/slam/map.txt.   The pose of the first marker detected is set such that the robot is at (0,0,0,0,0,0).  

Please see the [documentation](http://wiki.ros.org/fiducial_slam?distro=kinetic) for more details

There is also a [script](https://github.com/UbiquityRobotics/fiducials/blob/kinetic-devel/fiducial_slam/scripts/fit_plane.py) that can be used to test how planar the markers are.

closed 2019-04-04 by jim-v

innovatim, 2019-04-04:
@jim-v Hello,sir. Does 'map.txt' record the pose of each marker relative to the robot? Is there a way to simply get the pose of every marker relative to the original marker?(the first marker detected is set to origin marker and every value of the pose is 0). I don't need the location of the robot currently. I just want the pose of every marker.

jim-v, 2019-04-04:
The map file contains the pose of the fiducials relative to the map
co-ordinate system.  It is independent of where the robot is.

I'm not sure if by pose, you mean a 6 degree of freedom pose, or just the
x,y,z linear portion.   If you just want the linear portion, you can
subtract the position of the first fiducial from all the others.   If you
want a full 6 DOF pose, you need to create a matrix from the pose of the
first and take its inverse, then multiply all the other poses by that.

innovatim, 2019-04-04:
Can `map co-ordinate system` be understood as the position of the robot when the first marker is detected. In the subsequent robot movement process, the relative position of each detected marker and `map co-ordinate system` will be written to the file `map.txt`. 
    I just want to get the x, y and theta of each marker relative to the origin marker.
> 29 0.067829 -0.463809 2.534852 167.871289 -0.360139 -177.054608 4011196.711223 78 36
> 36 -0.171610 -0.094652 0.905192 -157.835704 -18.248970 24.028769 0.000000 0 29
    This is the content of the map.txt I ran `fiducial_slam` with ID 29 maker and ID 36 marker. According to your statement, 29 is the origin(because it's the first detected marker), then the plane coordinate of ID 36 is relative to 29 (-0.171610 - 0.067829, -0.094652 - 0.463809).
    And here is the schematic diagram of my thoughts(am I right?):[image]

jim-v, 2019-04-04:
The origin of the map coordinate system is where the robot was when the first marker was detected.  The first marker is not the origin, but it is treated differently to the others in that its position won't change.
The first marker detected was 36, this can be determined because the 8th value in that line is 0, meaning it's pose won't be updated.

In the example above, the markers do not seem to be in a plane, since their angles are different.   I recommend running the script I pointed you to to see if they are in fact in a plane, then you can convert to the 2D coordinates of the plane if that is what you are trying to do.

innovatim, 2019-04-06:
@jim-v Hello,sir. I just tested [fiducials](https://github.com/UbiquityRobotics/fiducials) with the following A4 paper with aruco markers printed. 
[image]
Obviously, they are all on one plane, I ran the [script](https://github.com/UbiquityRobotics/fiducials/blob/kinetic-devel/fiducial_slam/scripts/fit_plane.py) you mentioned, but the feedback seems to show that they are not on a plane:
[image]

jim-v, 2019-04-06:
The blue dots are the position of each marker and the red dots are that position projected onto the plane that is fitted.  If the blue and red dots line up, then there is a good fit.   I ran the script on the map data you sent and got this:

```
Plane normal: [-0.07823215  0.12024974  0.98965637]
slope: -4.519821 deg in X 6.927853 deg in Y
residual: 0.001599
```

The residual being low suggests they plane is a good fit.

You could use the `move_origin.py` to subtract the fist marker.  But the x,y,z coordinates for the markers would still be in the map frame, not relative to the plane.   I'm still not understanding what you are trying to do.

jim-v, 2019-04-06:
Does this do what you want:

```
def projectPointsOntoPlane(C, N, points):
    Xaxis = np.cross(N, (1, 0, 0))
    Yaxis = np.cross(N, (0, 1, 0))
    for point in points:
        p = point - N
        x = np.dot(p, Xaxis)
        y = np.dot(p, Yaxis)
        print(x, y)
```

On your data, it gives:
```
(-0.088385334142984495, 0.024801586942626054)
(-0.1807616392780485, 0.022125154945395853)
(-0.086159192309937474, -0.066798929266142018)
(-0.17828718856925246, -0.06969871294964286)
(-0.084177083902052641, -0.15869247573468631)
(-0.1761139744702209, -0.16101287887549695)
```
[image]

innovatim, 2019-04-07:
Thank you for your patience. What I want is the poses of all the aruco markers relative to a specified marker. The `map.txt` generated by `fiducial_slam` contains the poses of all markers relative to the `map co-ordinate system`. I am wondering if it is feasible to let `move_origin.py` read the `map.txt`, and then convert the poses of the markers relative to the `map co-ordinate system` to the poses of the markers relative to the specified marker.

jim-v, 2019-04-07:
Yes, you could use `move_origin.py` to subtract the translation of one pose from the others.  But note that it wouldn't convert the poses into the co-ordinate system of the specified marker, since the script only deals with translation, not rotation.

You could also manually create a map that puts which ever marker you want at the origin and then detect the others.