UbiquityRobotics/fiducials #172: Wrong pose estimation for Raspberry Pi + camera module v2.0

mc-12, 2019-05-17:
I'm currently working on a project which involves a small robot controlled by a Raspberry Pi which has to know it's position to perform a task. For that, I want to use aruco markers + the official camera module. I followed the instructions at https://github.com/UbiquityRobotics/raspicam_node/blob/kinetic/README.md to install the raspicam package and respectively http://wiki.ros.org/fiducials to install the fiducials package on the Pi. The markers were created using the command `rosrun aruco_detect create_markers.py 100 112 fiducials.pdf`.

My setup consists of the Pi only running the raspicam node and my laptop which runs roscore and the aruco detect node. I configured the ros master uri accordingly. Now I am using the following commands to get everything started:
On my laptop:
`roscore`
`roslaunch aruco_detect aruco_detect.launch camera:=/raspicam_node`

On the Pi:
`roslaunch raspicam_node camerav2_1280x960.launch`

The detection seems to work fine regarding the rectangle in the image as you can see here on the right side:
[image]

Unfortunately the translation coordinates are not correct. For example (see red rectangle in the image) the real z distance between the camera is more than twice as long as stated by the z coordinate. The other coordinates give some wrong values as well. As a result, I can't use the coordinates given by the aruco node to say anything about the camera position relative to a marker. Later on I also want to use the slam node, but first I should get this working, I guess...

Do I miss something important or may interpret these values wrong? Or might it be possible that the camera calibration file does not match the current camera module? Any hint or input is highly appreciated! Thanks :)

jim-v, 2019-05-17:
Hello,

The output you highlighted in red should be the translation vector between the fiducial and the camera.  If it is wildly off, it's worth checking that the fiducial is physically the size expected by `aruco_detect`.

ln-12, 2019-05-18:
Yep, I also checked that before. 

Today I tried out the calibration tutorial at https://github.com/UbiquityRobotics/raspicam_node/#calibration to get my own calibration file and now everything works as expected. I'm adding the content of the generated .yaml file below if someone has the same problem and finds this issue.

image_width: 1280
image_height: 960
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [2228.787723, 0.000000, 634.910206, 0.000000, 2239.174910, 393.851948, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.524763, 1.514570, -0.002089, 0.007610, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [2151.775146, 0.000000, 639.386937, 0.000000, 0.000000, 2186.162598, 389.994685, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

closed 2019-05-18 by ln-12

jim-v, 2019-05-18:
Thanks for the calibration data!