UbiquityRobotics/fiducials #178: fiducial_len_override causing a CV error

SpyGuyIan, 2019-06-11:
Upon setting fiducial_len_override, aruco_detect throws errors even after noting that it has detected all 4 markers in the image.

```
[INFO] [1560215349.612089574]: Detected 4 markers
OpenCV Error: Assertion failed (markerLength > 0) in getSingleMarkerObjectPoints, file /tmp/binarydeb/ros-kinetic-aruco-detect-0.11.0/src/aruco_detect.cpp, line 127
[ERROR] [1560215349.612601219]: cv exception: /tmp/binarydeb/ros-kinetic-aruco-detect-0.11.0/src/aruco_detect.cpp:127: error: (-215) markerLength > 0 in function getSingleMarkerObjectPoints
```

Here is the relevant part of my launch file:

```
<node pkg="aruco_detect" type="aruco_detect" name="aruco_detect" output="screen">
        <param name="fiducial_len" value="0.1"/>
        <param name="fiducial_len_override" value="0-3: 0.1, 4: 0.15"/>
        <param name="dictionary" value="0"/>
        <param name="publish_images" value="true"/>
        <remap from="/camera" to="/cv_camera/image_raw"/>
        <remap from="/camera_info" to="/cv_camera/camera_info"/>
</node>
```

My use of fiducial_len_override looks very similar to the documentation so I am confused where the error stems from.

jim-v, 2019-06-11:
Thanks for reporting the issue.  The floating point lengths are not being passed correctly, causing the length for fiducials 0-3 to be 0, which causes the assertion error.   We will make a fix shortly.

```
[ INFO] [1560238888.881680030]: Setting fiducial id range 0 - 3 length to 0.000000
[ INFO] [1560238888.882160282]: Setting fiducial id 4 length to 4.000000
```

rohbotics, 2019-06-11:
This should be fixed after #180 is merged in if you are building from source. Thanks again for the bug report.

PR #180 (2019-06-11): Fix fiducial_len_override bug
Fixes #178

Tested with:
```
        <param name="fiducial_len_override" value="0-3: 0.1, 4: 0.15"/>
```

Before:

```[ INFO] [1560238888.881680030]: Setting fiducial id range 0 - 3 length to 0.000000
[ INFO] [1560238888.882160282]: Setting fiducial id 4 length to 4.000000
```

After:

```[ INFO] [1560239394.526257919]: Setting fiducial id range 0 - 3 length to 0.100000
[ INFO] [1560239394.526717422]: Setting fiducial id 4 length to 0.150000
```
closed 2019-06-11 by jim-v