UbiquityRobotics/fiducials #183: Global topic names prevent usage in multi-robot scenarios

argoscoder, 2019-07-05:
Hello,
I am working on a cooperative multi-robot scenario, and I would like to use the aruco_detect node to get the transforms between each of the robots and some Aruco markers in order to plan cooperative manipulation tasks. However, when looking at the code, it seems that the topic names are defined globally, for example :
```c++
pose_pub = new ros::Publisher(nh.advertise<fiducial_msgs::FiducialTransformArray>("/fiducial_transforms", 1));
```
This results in both nodes publishing their results on the same topics, without a way of differentiating between the 2 (or more) robots using the node.
Is this forced by some other parts of the package, or could it be changed to local names (without the / ) to allow prefixing the topics with namespaces?

Thanks in advance!
Best,
Thibault

jim-v, 2019-07-05:
Hi Thibault,

We could change the names to not start with /.   If you are willing to work on this, please feel free to send a pull request.

Jim

rohbotics, 2019-07-08:
You could also write your own launch files that rename the topics or launch the nodes in a namespace.

argoscoder, 2019-07-08:
Hello,

Thanks for your answers! I had tried to run the node in a namespace, but the global naming was overriding the namespace prefix. I should have thought about remapping the topic though, this works as expected :)

The difference I see between using local names or remapping the topics to local names directly in the launch file is only that the fisrt solution resolves to the topic name being "/robot_ns/aruco_detect/fiducial_transforms" (for example) while the 2nd one resolves to "/robot_ns/fiducial_transforms". 
I think this issue can be closed since the workaround is quite easy
Thibault

rohbotics, 2019-07-09:
I think the right thing to do is to make the topics published from a public
node handle, without a leading /.
That way namespacing should work by default, even without remapping.

Rohan

rohbotics, 2019-07-13:
I made a PR that should solve this in the correct way #185.

Thanks for your bug report!

closed 2019-07-20 by rohbotics