UbiquityRobotics/fiducials #187: Fiducial_slam not building from catkin_make command

ArghyaChatterjee, 2019-07-14:
I tried several times but it seems that this file is never going to build from source. What should I do now?
Note that I also tried with the hector slum model and it's unable to compile it as well. Looks like slam is never going to run in my raspberry pi. I am using ubiquity robotics latest version of Ubuntu Mate (16.04) as my OS of raspberry pi where ros kinetic is preinstalled.
[image]

jim-v, 2019-07-14:
If you are building on a Raspberry Pi, it could be running out of memory.  I usually do `catkin_make -j1` to just run one command at a time.

ArghyaChatterjee, 2019-07-14:
If I try to do it this way, catkin_make --pkg fiducial_slam, then it hangs exactly in the same file which is shown in the screenshot (fiducial_slam.cpp.o). I can't even move the cursor. I had to unplug the battery cable and make pi boot again after 10-15 minutes of pause.

jim-v, 2019-07-14:
It will be compiling two files at the same time unless you specify `-j1` and will run out of memory.

ArghyaChatterjee, 2019-07-14:
Thanks jim. You helped me save a lot of time. It worked. I am closing this issue here.

closed 2019-07-14 by ArghyaChatterjee