UbiquityRobotics/fiducials #188: No image showing up in rviz

ArghyaChatterjee, 2019-07-16:
I calibrated my camera using this tutorial.
https://github.com/UbiquityRobotics/raspicam_node
I built the package from source following this tutorial.
https://github.com/UbiquityRobotics/fiducials
I tried to follow this tutorial from ROS official website. 
http://wiki.ros.org/fiducials
First 2 command lines perform good and the last one also performs fine but when rviz shows up, no image is found. 
[image]
I am new to ROS. I am running it on a raspberry pi. No visible error shows except DRI2: failed to authenticate.

rohbotics, 2019-07-17:
Hi, Thanks for reporting your issue.

Can you first make sure that you are getting camera data, by changing the Image topic to `/raspicam_node/image`?

Wanja1204, 2019-09-01:
hello, i have the same problem, when I changing the image topic to image raw for my webcam i get a picture but no marker recognition

jim-v, 2019-09-01:
Hi,

You should see the camera image without annotations on the topic `/raspicam_node/image` and with them on `/raspicam_node/image`, assuming the `aruco_detect` node is running.

 If this is not the case, please let us know and we will figure out why.

yulan0215, 2020-08-06:
> I calibrated my camera using this tutorial.
> https://github.com/UbiquityRobotics/raspicam_node
> I built the package from source following this tutorial.
> https://github.com/UbiquityRobotics/fiducials
> I [...]

Did you solve this problem? How did you solve this problem? Thank you!

madgrizzle, 2020-10-05:
@yulan0215, not clear what problems you may be having, but you may want to check this issue and see if its because aruco_detect isn't subscribed to the camera properly.

https://github.com/UbiquityRobotics/fiducials/issues/228

lewsmith, 2023-06-13:
Anybody know how to solve this issue? I have a usb cam and I can see the image in rviz for the topic `/camera/image_raw` but nothing for the topic `/fiducial_images`

anfederman, 2023-06-14:
Please send listing from:

rostopic list

You should see raspicam_node/image and fiducial_images/compressed.

anfederman, 2023-06-14:
I think you need the compressed image, but please send the complete
rostopic list.

And then see if rostopic echo fiducial_images is sending data