UbiquityRobotics/fiducials #190: In fiducial slam package got Oscillation in fiducial_pose.

vishalsleotechsa, 2019-07-23:
I am using the UbiquityRobotics/fiducials package, I have followed all the step mention in readme.md file, after using this package I am getting an error as the fiducial_pose(robot pose in map frame) is oscillating continuously while robot and markers are in a stationary condition. so can anyone will give the solution for that! 
 
I am using the following specifications,

ROS distribution: indigo
camera: Kinect v2
camera is calibrated 
camera front-facing
marker on the wall

jim-v, 2019-07-23:
There are usually some small oscillations in the robot pose from fiducial_slam.  Can you record a bag file
using the command `rosbag record -a` and send it to us so we can look into it and see if there is something else going on here.

vishalsleotechsa, 2019-07-24:
thank you so much
please find the attachments and data.
here is the fiducial pose data
header: 
  seq: 229
  stamp: 
    secs: 1563952902
    nsecs: 811744871
  frame_id: map
pose: 
  pose: 
    position: 
      **x: 0.827669586504
      y: 0.195369691712**
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.991645769297
      w: 0.128990961834
  covariance: [0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01]
---
header: 
  seq: 230
  stamp: 
    secs: 1563952903
    nsecs: 120705026
  frame_id: map
pose: 
  pose: 
    position: 
      **x: 0.7978947861
      y: 0.0429898257556**
      z: 0.0
    orientation: 
      x: 0.0
[fiducial_pose_1_2019-07-24-12-51-40.bag.zip](https://github.com/UbiquityRobotics/fiducials/files/3425745/fiducial_pose_1_2019-07-24-12-51-40.bag.zip)

      y: 0.0
      z: -0.995863777578
      w: 0.090858882389
  covariance: [0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01]
---

it's oscillating by a large distance is fiducial_pose here robot and markers are in a stationary condition.
I have attached the bag file of the topics of /fiducial_pose /fiducial_detect/fiducial_transforms.
Also, we use the front-facing camera for that we use the static transformation
which is 

 " **<node pkg="tf" type="static_transform_publisher" name="base_link_to_raspi_camera_frame" args="0.2300 -0.094 0.32 -1.57079633 0 -1.57079633  base_link raspi_camera_frame 100"/>**  "

and initially, we use tf in the map.txt file  which is 
" 1 -1.22 0.0 0.1681 90.00000000000006 -6.3611093629270335e-15 81.99999999999999 0.0 1 "

[VID_20190724_125157.mp4.zip](https://github.com/UbiquityRobotics/fiducials/files/3425782/VID_20190724_125157.mp4.zip)

vishalsleotechsa, 2019-07-24:
**<node pkg="tf" type="static_transform_publisher" name="base_link_to_raspi_camera_frame" args="0.2300 -0.094 0.32 -1.57079633 0 -1.57079633  base_link raspi_camera_frame 100"/>**

jim-v, 2019-07-24:
Hi,

I haven't had a chance to look at the bag file yet, but one potential problem is that two transforms from base_link to the camera are being published.   Instead of the static transform publisher, the way to change from upward to forward camera is to edit `/etc/ubiquity/robot.yaml` and make sure you have

```
raspicam:
    position: 'forward'

```

vishalsleotechsa, 2019-07-25:
hi,

I couldn't get the  " /etc/ubiquity/robot.yaml " file, I have download the fiducial package using this link 
"https://github.com/UbiquityRobotics/fiducials.git'

for ros indigo distribution and compile and build the "fiducials-indigo-devel"
package.

jim-v, 2019-07-25:
My mistake, I thought that you may have been using the Ubiquity Robotics Raspberry Pi interface.
I will take a look at the bag file at the weekend.

jim-v, 2019-07-28:
I looked at the bag file you sent and from the message types, it looks like you are running code from the Indigo branch.  I also just noticed that you mentioned it in your previous message.  That code hasn't been maintained for several years and we are going to remove it eventually, since Indigo support ended in April.  The kinetic-devel branch has many improvements (including a rewrite of the Python fiducial_slam in C++) and is known to run on Kinetic and Melodic.  What version of ROS are you running?

vishalsleotechsa, 2019-08-02:
sorry for let,
we are run the ROS-Indigo version.
we are trying run the package on the kinetic-devel branch.
thank you.

vishalsleotechsa, 2019-08-21:
I apologize for the delay, 
I have run the fiducial slam node on ROS kinetic distribution. 
in that section still, the robot position gets oscillating for the static position of robot and marker, when markers are located on the wall. 
I am using the following specifications,

ROS distribution: kinetic
camera: Kinect v2
camera is calibrated
camera front-facing
marker on the wall

video of robot postion on rviz
[VID_20190821_150842.mp4.zip](https://github.com/UbiquityRobotics/fiducials/files/3525424/VID_20190821_150842.mp4.zip)

jim-v, 2019-08-25:
Some jitter in the order of 1-2 cm is expected, but what you have indicates something is wrong.   Things to check are

- The camera is well calibrated
- The size of the marker is as specified in the launch file
- The markers are flat
- The transform for the base -> camera is correct
- Odometry is being correctly published, or not used by fiducial_slam

If you can create another bag file of `/fiducial_transforms` I can try to reproduce the problem.

vishalsleotechsa, 2019-08-26:
thank you for the response,
we check all topics as you mention and create the bag file and inform to you.

vishalsleotechsa, 2019-08-29:
hello,

I have check all the aspects of you mention,
1. camera calibration,
  we do the camera calibration, we use the Kinect camera use the following link for the camera  calibration  
" https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration "
2. size of the marker
    yes, we specified the size of the marker in the launch file.
3. The markers are flat