UbiquityRobotics/fiducials #191: Is there a minimum detection distance?

canberkgurel, 2019-08-02:
I noticed that the aruco markers are no longer detected when the camera is any closer than ~40 cm. Is this a known feature/issue? Note that when the camera is within the 10-40 cm range, the entire marker and the surrounding white border is still visible.

I know that the minimum white border recommended is slightly larger than 1 square of the marker. However, this is not the case for my markers, could this be playing a role here?

I am using is the Intel RealSense R200 camera from Intel. The camera is calibrated.

jim-v, 2019-08-04:
Hi,

I suspect the `maxMarkerPerimeterRate` parameter is the one you should change.
The parameters `aruco_detect` node uses can be dynamically configured.   You can do this by running `rosrun rqt_reconfigure rqt_reconfigure` while the node is running to find optimal values.  These values can then be set from the launch file.

Jim

rohbotics, 2020-02-10:
Closing this due to age and lack of response. 

@canberkgurel  Feel free to re-open if you still have questions.

closed 2020-02-10 by rohbotics