UbiquityRobotics/fiducials #192: Intermittent Tag Detection under Seemingly Good Conditions Outdoors

m1baldwin, 2019-08-16:
I have been using the aruco_detect package with a realsense r200 camera with a uav to provide a relative pose estimate between the two to aid navigation. In our lab, with the exact same mounting of the camera (and a smaller tag) we are seeing stable detection of the tag and generation of the pose estimate. However, when we move outdoors and test (with a larger tag), we see intermittent detection, even when the visual of the tag seems to be quite good. I am attaching a screen shot of one such situation where we captured a frame in image_view, and also ran rostopic echo on the /fiducial_transforms topic; what you can see is even though the tag is in clear view, there are multiple frames of no detection. 

[image]

We have tried changing the brightness/contrast of the realsense using v4l cli commands, and though we noticed a difference in the video feed, we still saw poor detection, even if we darkened the images to the point that the tag was really the only noticeable thing. 

I am wondering if this could be due to too much excessive motion of the drone during its hover (vs. the lab where it is stable under the correction of an external vision system) or if this is due to something else in the environment outdoors (intense sunlight, uneven lighting, etc.)
[image]

canberkgurel, 2019-08-20:
@jim2v Any thoughts on this? Thanks!

jim-v, 2019-08-21:
There are some parameters in [aruco_detect](http://wiki.ros.org/aruco_detect?distro=melodicl) that limit the size of markers that will be detected relative to the size of the image, for example `maxMarkerPerimeterRate`.  My suggestion is to use dynamic reconfigure to find settings that work and then apply them through the launch file.