UbiquityRobotics/fiducials #196: Aruco detect not publishing anything

ashwinv96, 2019-10-28:
Hi, I followed the calibration steps and finally hit 'commit' to save the changes on my raspicam v2. My camera is publishing values and I can read them back using rostopic. The camera also works when using rviz.

However, I can not get any detection of aruco markers even when I am running both the 'aruco_detect' and 'fiducial_slam' nodes. I used the below commands.
roslaunch aruco_detect aruco_detect.launch
roslaunch fiducial_slam fiducial_slam.launch

Please see the attached image. The aruco and fiducial slam nodes state that they are ready for detection but even with markers placed right in front of the camera, no activity takes place. What can I do? Thanks!
[image]

rohbotics, 2019-10-28:
When you launch aruco_detect, you need to specify the camera name `roslaunch aruco_detect aruco_detect camera:=/raspicam_node` and `roslaunch fiducial_slam fiducial_slam camera:=/raspicam_node`. 

This should do what you want :)

Let me know if that works, or if u run into anymore issues.

ashwinv96, 2019-10-28:
@rohbotics that fixed my issue! thank you! 

FYI, I got it to work after adding '.launch' to the second instance of the node name in both the lines you posted. 
So:
roslaunch aruco_detect aruco_detect.launch camera:=/raspicam_node 
roslaunch fiducial_slam fiducial_slam.launch camera:=/raspicam_node

closed 2019-10-28 by ashwinv96

rohbotics, 2019-10-28:
Awesome, thanks for catching the typo too.

Good luck with your project :+1: 

Consider tipping us a couple dollars to support the ongoing development of all our opensource code. There is a donation button on our download page https://downloads.ubiquityrobotics.com/pi.html, or you could simply PayPal to tips@ubiqityrobotics.com. Thanks!