UbiquityRobotics/fiducials #203: Hard Coded Image Topic

danem, 2020-01-15:
Unless I'm mistaken, the image topic subscribed to in aruco_detect is hard coded to "camera" set [here](https://github.com/UbiquityRobotics/fiducials/blob/kinetic-devel/aruco_detect/src/aruco_detect.cpp#L568). Shouldn't this be exposed as a launch file parameter?

rohbotics, 2020-01-15:
You can use a remap to change the topic like we do in the launch file: https://github.com/UbiquityRobotics/fiducials/blob/kinetic-devel/aruco_detect/launch/aruco_detect.launch#L22 Because remapping is an option, we decided not to provide a parameter as well.

danem, 2020-01-15:
Ok, I was hoping for a way to simply do it via roslaunch (I'm not aware of a way to perform remapping via the roslaunch CLI). This works though.

Thanks

closed 2020-01-15 by danem