UbiquityRobotics/fiducials #204: Incorrect pose is published if odomFrame baseFrame lookup fails

rcywongaa, 2020-01-20:
I'm referring to the following snippet of code from [map.cpp](https://github.com/UbiquityRobotics/fiducials/blob/745ae67ab3f703013a0e1562c85bd3dd819ef304/fiducial_slam/src/map.cpp#L352)
```
    tf2::Stamped<TransformWithVariance> outPose = basePose;
    outPose.frame_id_ = mapFrame;
    std::string outFrame = baseFrame;

    if (!odomFrame.empty()) {
        outFrame = odomFrame;
        tf2::Transform odomTransform;
        if (lookupTransform(odomFrame, baseFrame, outPose.stamp_, odomTransform)) {
            outPose.setData(basePose * odomTransform.inverse());
            outFrame = odomFrame;

            tf2::Vector3 c = odomTransform.getOrigin();
            ROS_INFO("odom   %lf %lf %lf", c.x(), c.y(), c.z());
        }
    }
    ...
    poseTf = toMsg(outPose);
    poseTf.child_frame_id = outFrame;
```
If `!odomFrame.empty()` and the `lookupTransform` fails, the resultant `poseTf` that gets published is a Frankenstein TF where it claims to be from map to odom but in fact it is from map to base.

I believe the correct behavior should be to not publish anything in the event that `lookupTransform` fails

As an aside, is it wise to have zero timeout for [`lookupTransform`](https://github.com/UbiquityRobotics/fiducials/blob/745ae67ab3f703013a0e1562c85bd3dd819ef304/fiducial_slam/src/map.cpp#L235)?

jim-v, 2020-01-20:
Hi Rufus,

Thanks for the feedback.    I agree that it would be better to not publish anything in the event that the Tf lookup fails.

However, I don't understand your comment about the timeout.  What we are doing is looking up the transform at the time contained in `basePose`

jim-v, 2020-01-20:
Actually, the Tf published is not going to be Frankenstien in either case. `outFrame` is intitialized to `baseFrame` and only changed to `odomFrame` if the lookup is successful.

rcywongaa, 2020-01-20:
Hi Jim,

It's probably an isolated case which can be ignored (maybe my computer just isn't strong enough to handle all the TF publishing).  I notice that the `lookupTransform` fails almost 30% of the time, adding a short timeout of 0.5s to the lookup solved it for me.

`outFrame` is set to `odomFrame` immediately after the `if (!odomFrame.empty())`.  In fact, it is set again when the lookup succeeds.

In any case, I believe it should not publish `map->base_frame` when user expects it to only publish `map->odom` (as indicated by providing the `odom` parameter)

jim-v, 2020-01-20:
Hi Rufus,

Thanks for the suggestions

rohbotics, 2020-02-10:
Closing because the PR was merged.

closed 2020-02-10 by rohbotics