UbiquityRobotics/fiducials #208: cv_bridge exception: [8UC3] is not a color format. but [bgr8] is. The conversion does not make sense

Hackerman342, 2020-02-09:
Hi,

I am changing hardware from a logitech webcam to an Arducam (https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield). When running aruco_detect with the Arducam, I get this error:

 cv_bridge exception: [8UC3] is not a color format. but [bgr8] is. The conversion does not make sense.

Any suggestions on how to adjust aruco_detect or what node I could add to receive/convert this [8UC3] image type for use with aruco_detect?

Thank you!

Hackerman342, 2020-02-09:
I found a way to change the image being published to bgr8. Please delete this issue.
Thanks.

rohbotics, 2020-02-09:
Hey @Hackerman342,

Glad that you resolved your issue. Instead of deleting this post, it would be great if you could post your solution in case someone stumbles across this in the future.

Rohan

Hackerman342, 2020-02-10:
Of course. In order to publish the image as bgr8, I changed a line in the image publisher node (python) from:
            img_msg = bridge.cv2_to_imgmsg(image)
to:            
            img_msg = bridge.cv2_to_imgmsg(image, "bgr8")

Hackerman342, 2020-02-11:
Hi Rohan,

Sorry, I closed this issue too early. Now aruco_detect receives the stream from the Arducam, processes it, and returns a /fiducial_images video stream, but does not detect any markers when they are in the display. The aruco_detect configuration is the same as when it worked before with the logitech webcam, besides image topic name changes in the launch file. I have played around a bit and can't figure out why, but my best guess is that the image channels are now labelled as what aruco_detect expects, but are not truly in that format. I've included a couple bagfiles (one recorded through a rosmaster connection (laggy) and one recorded locally (better)) with relevant launch files and scripts in the following drive location. Please let me know if you have any suggestions or if any additional information would help debug.

https://drive.google.com/drive/folders/1WHhA3K0tS7YYoXKSBcGXIKKpAuPy2Tdm?usp=sharing

Thanks for the help!
Kyle

Hackerman342, 2020-02-13:
Hi again,
Is it possible that whites in the image from this new camera is too offwhite? See attached image comparing the new arducam raw_image (top) to the fiducial images output using the old logitech webcam (bottom). Let me know if you have any recommendations on testing / resolving this.

Thanks,
Kyle

[image]

closed 2020-02-13 by Hackerman342

rohbotics, 2020-02-16:
Glad you solved the issue Kyle!

The non-mirror symmetry is quite useful in rooms that have actual mirrors in them :), but is more annoying when you don't realize that your camera is mirrored.

Good luck on your project.