UbiquityRobotics/fiducials #213: How can I interpretate object_error field as error of distance measurment to marker?

MoscowskyAnton, 2020-03-25:
Hello!
I wonder how can I use object_error parameter as accuracy of distance to maker.
I run my robot in Gazebo, where it moved many various directions and obtained data. 
[image]
[image]
Then I plot on one figure error between distance to marker from gazebo and solved by aruco_detect node (as len of translation vector), and appropriate to it object_error.
Here what i get
[image]
Seems there some correlation between it, but how to transform one from another correctly?
Thanks!

rohbotics, 2020-04-14:
Hey @MoscowskyAnton 
Sorry for the late response to this. 

The object error is a heuristic estimate of the error in the marker position based on the reprojection error. The reprojection error is calculated by seeing how well the estimated marker position aligns with the measured vertices of the camera. Obviously this fails to completely account for all possible errors in the system, so it will likely always be less than the real error of the system.

Some large factors in the error of the system are:

- Quantization error from the camera
- Poor camera calibration
- Systematic or random error in vertex estimation partially resulting from the above 2

Because when solving from the vertices to a transform, some of error in vertex estimation can prevent an exact match. The estimated vertices may be impossible to construct into a square, so we approximate it. This approximation is what results in reprojection error.

Let's say you had an error source that caused the vertices to always be estimated 1 pixel to the left of their actual positions. This would likely have no effect on reprojection error, even though it would obviously cause real error.

If you have a better way to estimate the actual error I am open to hear it :)

rohbotics, 2020-04-30:
Hey @MoscowskyAnton, I am closing this for now. Feel free to re-open if you have any further questions.

closed 2020-04-30 by rohbotics