UbiquityRobotics/fiducials #215: Is there any way to start fiducial slam without any prior map information every time.

JanezCim, 2020-04-14:
I am wondering if there is any way to launch fiducial slam without any prior map of markers every time, so to speak call service /fiducial_slam/clear_map on every launch.

I've tried calling service from launch file with: 
<node pkg="rosservice" type="rosservice" name="clear_fiducial_map" args="call --wait /fiducial_slam/clear_map" />
But somehow it does not work.

rohbotics, 2020-04-14:
We don't have anything that is directly intended for this, but specifying an empty file for the `initial_map_file` parameter would have the effect you want.

JanezCim, 2020-04-16:
Yes, that does seem to be working, thank you!

rohbotics, 2020-04-30:
Closing this because we found a workaround solution.

closed 2020-04-30 by rohbotics