UbiquityRobotics/fiducials #216: Fiducial slam does not map or estimate tf on markers with larger id then 107

JanezCim, 2020-04-17:
[image]

Is there a parameter that I am missing where I could indicate that I want it to do slam on other markers then just 100-107?

rohbotics, 2020-04-17:
Hi @JanezCim 

Based on the image, it looks like aruco detect isn't even detecting those markers. What launch file are you using? Did you print all the markers with the script in aruco detect?

mjstn, 2020-04-17:
Adding two cents.   Be sure your launch files do not specify any fiducials to ignore which is unlikely but look for at launch startup the word 'ignore' and also if fiducials are in the ignoreIds list a ROS log should show truely ignored ids.   If they don't show up in runtime ROS_INFO messages saying  'Ignoring id 123' then the issue may be they are not even being detected as Rohan is trying to diagnose which sounds like the real issue.   Also check when the launch starts out can you look for the parameters it prints right at the start and see if it has  fiducial_len as the value you are using which for ceiling is normally 0.14 meters for our default.   This is unrelated but always good to know the actual value of your markers is what is in use.     Also be sure your focus is good and I see in your picture to lower right a large square of dark shadow is covering one fiducial.  See if things get seen when the fiducials are in even and good lighting and focus of camera is good.

JanezCim, 2020-04-18:
I see what happened, because of the shadow on the aruco marker (seen in the camera view), the two markers 107 and 108 could not have been detected together in one frame. And because fiducial slam actually needs to see two markers at the same time to map the new one, it did not map. 

I am working with a BAG file, and that is why I always got the same outcome. Nevermind, sorry for the confusion. Thank you for the quick reply!

closed 2020-04-18 by JanezCim