UbiquityRobotics/fiducials #217: Investigate SolvePNP vs SolvePNPRansac

rohbotics, 2020-04-20:
Looking at the aruco_detect code, it is using SolvePNP to go from the image points to the object points. In the OpenCV docs there is also a function called SolvePNPRansac which claims to more more resistant to outliers. Would this be a good idea to use? Or would it be useless as we only have 4 points anyway?

https://docs.opencv.org/3.2.0/d9/d0c/group__calib3d.html#ga50620f0e26e02caa2e9adc07b5fbf24e

@jim-v Maybe you can give some thoughts on this.

jim-v, 2020-04-20:
My intuition is that a RANSAC approach would not be useful given we only have four points.  If only three of them match well, it is not a good pose estimate.

rohbotics, 2020-04-20:
Yes that makes sense. If we revive multi-fiducial with a read-only map it might be useful to use the ransac mode.

closed 2020-04-20 by rohbotics