UbiquityRobotics/fiducials #219: Fiducial_slam not subscribing to /fiducial_transform

RegGrobler, 2020-06-09:
Good day

I am using robot upstart to run my launch file when my jetson nano boots up. The launch file runs correctly when run from the terminal using roslaunch but not when starting the service created by robot_upstart.

Both aruco detect and fiducial_slam nodes are being launched but fiducial_slam is unable to subscribe to the fiducial_transform topic. This appears to be my only issue. Again this does not happen when running launch file manually from terminal. 

[image]

rqt_graph does not show the fiducial_transform topic being published by aruco_detect node however I can confirm that this topic is being published correctly using rostopic echo. 

I have had similar issues in the past with using robot_upstart but these issues were permission related. I cannot think of any special permissions not set for this application.

Any help would be greatly appreciated.

Here is my launch file:

[image]

Here is the launch file for fiducial_slam:
[image]

rohbotics, 2020-06-11:
Hi @RegGrobler,

Interesting that this only happens under robot_upstart. What is the output of `rostopic info /fiducial_transform`?

It doesn't look like fiducial slam is running, can you verify it has the permissions it needs to read and write to ~/.ros/slam/map.txt for the user that it is running as?

RegGrobler, 2020-06-12:
Hi @rohbotics 

I believe you are correct. This is most likely a permission issue as robot_upstart runs the launch file as an unprivileged user. 

I ran the command "chmod ugo+rwx map.txt". and can confirm that the permissions were set accordingly. This should give all users read write and execute privileges. This however did not work and im still stuck. Can you think of any other file permissions that need to be set?

RegGrobler, 2020-06-12:
Does fiducial_slam require any special group membership? 
Such as: adm, dialout, cdrom, sudo, audio, dip, video, plugdev, i2c, lpadmin, gdm, sambashare, weston-launch, gpio. 

Im asking because when robot_upstart launches the launch file, it loses all group memberships of the user it is launched as. Which in my case is the default user. 
Aruco_detect launches perfectly and I cannot think of any groups fiducial_slam would require that aruco_detect doesnt.

RegGrobler, 2020-06-12:
Output of rostopic info:

Type: fiducial_msgs/FiducialTransformArray

Publishers: 
 * /aruco_detect (http://Hanabi-Nano:34555/)

Subscribers: None

rohbotics, 2020-06-12:
I don't think fiducial_slam needs any of those groups, but it needs to be able to read write delete and create files in `$HOME/.ros/slam/map.txt`. If you are running as another user, make sure that HOME is set and somewhere with those permissions. Alternatively, you can specify a different map file location that is more convenient using the `map_file` rosparam.

rohbotics, 2020-07-10:
Closing this because of lack of response. If you are still have trouble, feel free to re-open.

closed 2020-07-10 by rohbotics