UbiquityRobotics/fiducials #226: CPU usage too high

Rojekk, 2020-08-24:
Hi,

I'm using aruco_detect package but it works at 30hz rate and consumes CPU more than %100, I need it to be %20,25 at most. I used ros::Rate and  rate.sleep() functionalities in main function and ran the program at 5hz and i got %60 CPU usage.Then i noticed the ciritical work is being done at imageCallback function so i did put rate to that function and now i get %40 CPU usage but that's still not enough for me. Is there a parameter for this that i couldn't find ? If not, how can i accomplish this ? 

Thank you

MoffKalast, 2020-08-24:
Well the way it works is that it should match the frequency of the data being sent, which would be the framerate of the camera. I think you should be able to reduce it with the throttle functionality of the topic_tools package.

There is something else to consider though, with 4 cores a raspberry pi technically has 400% max cpu usage, so it's only using about a quarter of all power in your case anyway.

Rojekk, 2020-08-24:
Actually i don't want to manipulate the topics, the critic line in the code is:

aruco::detectMarkers(cv_ptr->image, dictionary, corners, ids, detectorParams);

I think i just need to put a if(a timer returns boolean every determined period) before it. Do you think this will work and is there a functionality that does this ?

MoffKalast, 2020-08-24:
You could just as easily create a new topic with a lower frequency and have aruco read from that one, but I suppose it does add overhead. I suppose the easiest way would be to just add a boolean that gets toggled every message and only process it when it's set to true in the callback function, which would reduce the frequency to 15hz.

Rojekk, 2020-08-24:
There is already a system call which enables and disables the reading process, i can shut it down and cpu usage reduces to %2, problem is lowering the usage when it's on. So i need a timer which sets a boolean variable as true at for ex every 0,2 seconds other times it will be false. Then i will simply do 

if(boolean)
aruco::detectMarkers(cv_ptr->image, dictionary, corners, ids, detectorParams); 
So reading process will be done every 0,2 seconds only, that's my idea.
As i've seen Ros::Timer works with callback functions' periods and i don't want to touch callback function's rate. Do you know a simple way to accomplish this. 

Btw: I'm working with a asus mini pc (6 cores) and intel realsense camera.

Rojekk, 2020-08-26:
I solved the issue by ros::Timer. I generated a dynamic programHz parameter and timer object gets it's duration as 1/programHz, so if i use 5hz duration to wait becomes 0,2 seconds and whenever it is triggered it makes a boolean variable true and ar code detection is done, after that boolean variable becomes false till other callback happens after 0,2 seconds. With these i got %20-30 CPU usage.

closed 2020-08-26 by Rojekk