UbiquityRobotics/fiducials #238: Release aruco_detect for ROS noetic?

ajithcodesit, 2021-02-26:
Is it possible to release `aruco_detect` for ROS noetic?

MoffKalast, 2021-02-26:
As in an apt package? Should be possible. Does it compile on noetic right now? The marker gen script would probably need to be updated to python 3 I suppose.

ajithcodesit, 2021-02-26:
@MoffKalast  Yes as an apt package. I am able to compile the package in noetic but have not tested it. I was able to fix the marker generator script for python 3.

rohbotics, 2021-02-26:
@ajithcodesit If you could make a PR with any noetic required changes once you have tested that everything works that would be appreciated!

anfederman, 2021-02-28:
I noticed in aruco_gazebo there was a call to python2 in the !#
I think it made on the RPI4 both ARM32 and ARM64, also on my ubuntu 18,04/melodic laptop.

I just tried a catkin_make on AMD64 20.04 Noetic, make fails on aruco_detect.cpp on line 430,58. the error is about no matching function call to 'FiducialsNode::estimatePose ....
   ...
430 |

anfederman, 2021-02-28:
430  --- reprojectionerror);
                                       ^

closed 2021-02-28 by anfederman

anfederman, 2021-02-28:
This might be due to a cv2 version issue on line 222 it says candidate expects 6 arguments, 8 provided.