UbiquityRobotics/fiducials #239: Compilation error on melodic

JanezCim, 2021-03-01:
when trying to compile with catkin_make i get

```
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp: In member function ‘void FiducialsNode::poseEstimateCallback(const FiducialArrayConstPtr&)’:
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:430:58: error: no matching function for call to ‘FiducialsNode::estimatePoseSingleMarkers(std::vector<int>&, std::vector<std::vector<cv::Point_<float> > >&, float, cv::Mat&, cv::MatExpr, std::vector<cv::Vec<double, 3> >&, std::vector<cv::Vec<double, 3> >&, std::vector<double>&)’
                                         reprojectionError);
                                                          ^
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:222:6: note: candidate: void FiducialsNode::estimatePoseSingleMarkers(float, const cv::Mat&, const cv::Mat&, std::vector<cv::Vec<double, 3> >&, std::vector<cv::Vec<double, 3> >&, std::vector<double>&)
 void FiducialsNode::estimatePoseSingleMarkers(float markerLength,
      ^~~~~~~~~~~~~
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:222:6: note:   candidate expects 6 arguments, 8 provided
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:436:58: error: no matching function for call to ‘FiducialsNode::estimatePoseSingleMarkers(std::vector<int>&, std::vector<std::vector<cv::Point_<float> > >&, float, cv::Mat&, cv::Mat&, std::vector<cv::Vec<double, 3> >&, std::vector<cv::Vec<double, 3> >&, std::vector<double>&)’
                                         reprojectionError);
                                                          ^
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:222:6: note: candidate: void FiducialsNode::estimatePoseSingleMarkers(float, const cv::Mat&, const cv::Mat&, std::vector<cv::Vec<double, 3> >&, std::vector<cv::Vec<double, 3> >&, std::vector<double>&)
 void FiducialsNode::estimatePoseSingleMarkers(float markerLength,
      ^~~~~~~~~~~~~
/home/janez/Documents/Projects/ubiquityrobotics/qualcomm_ws/src/fiducials/aruco_detect/src/aruco_detect.cpp:222:6: note:   candidate expects 6 arguments, 8 provided
fiducials/aruco_detect/CMakeFiles/aruco_detect.dir/build.make:62: recipe for target 'fiducials/aruco_detect/CMakeFiles/aruco_detect.dir/src/aruco_detect.cpp.o' failed
make[2]: *** [fiducials/aruco_detect/CMakeFiles/aruco_detect.dir/src/aruco_detect.cpp.o] Error 1
CMakeFiles/Makefile2:8756: recipe for target 'fiducials/aruco_detect/CMakeFiles/aruco_detect.dir/all' failed
make[1]: *** [fiducials/aruco_detect/CMakeFiles/aruco_detect.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```

I think this is since last merge with lama branch. Please fix

MoffKalast, 2021-03-01:
This is the fiducials repo, what's this got to do with lama? What?

MoffKalast, 2021-03-01:
Alright whatever, I'm reverting that merge for now and we can put a fix and re-merge into the backlog for a day when someone actually has the time to deal with this issue.

MoffKalast, 2021-03-01:
@JanezCim Can you check if it works now?

JanezCim, 2021-03-01:
Aha, sorry for wrongly accusing teo then. Now it compiles alright

closed 2021-03-01 by JanezCim