UbiquityRobotics/fiducials #242: Aruco_detect.cpp make fails in Noetic

anfederman, 2021-03-02:
I compiled OK in Melodic with python2, opencv 3.3

It failed in catkin_make with python3, opencv 4.  

I think the same issue is in other fiducials nodes.  the function has a different number of variables in different versions of opencv,

rohbotics, 2021-04-08:
Can you post the exact error in here Alan? Thanks

anfederman, 2021-04-08:
This now works, I don't recall what the issue was or what I did to fix it, was probably a typo when I was looking at the code.  Aruco detect works, and fiducial_slam is showing markers in rviz in the correct locations, but I haven't tried building a map or navigating the robot based on fiducial_slam.

fiducial_follow is not working yet, I raised an issue in magni_demos.

not-vibhu, 2021-06-08:
Hi @anfederman 
Were you able to build this in Noetic?

igor-markovic, 2021-06-18:
I can confirm that the code from the kinetic-devel branch builds fine on a fresh install on Ubuntu 20.04 with ros-noetic-desktop-full installed, OpenCV version 4.2.

rohbotics, 2021-07-19:
Closing because the noetic-devel branch builds under noetic.

closed 2021-07-19 by rohbotics