UbiquityRobotics/fiducials #245: How can aruco-boards be detected with this package?

fmros, 2021-03-24:


JanezCim, 2021-03-25:
The instructions of how to use this package can be found on http://wiki.ros.org/fiducials. Or do you have more specific questions?

caioaamaral, 2021-03-25:
> > The instructions of how to use these packages can be found on http://wiki.ros.org/fiducials. Or do you have more specific questions?
> 
> I wanted to try to increase detection accuracy by using marker-boards(see this [link](https://stackoverflow.com/a/51736875)). So my question is, is it possible to detect a marker-board or aruco-board with this package?
> 
> See also here: https://docs.opencv.org/master/db/da9/tutorial_aruco_board_detection.html

Aruco detector will detect each tag displayed on your aruco board. If you're interested in a specific position, you will need to reference it (as a static tf for example) to the position of each tag (the detected position of course) and then use some sort of filter (kalman filter from robot_localization is a good example) to retrieve the optimal position estimation

fmros, 2021-03-26:
> Aruco detector will detect each tag displayed on your aruco board. If you're interested in a specific position, you will need to reference it (as a static tf for example) to the position of each tag [...]

So If i would want to get one position out of a board of 4 markers, I have to deal with the 4 detected fiducial frames of this package, and combine them to a group estimate with other techniques. Is this what you intend?

JanezCim, 2021-03-26:
> Aruco detector will detect each tag displayed on your aruco board. If you're interested in a specific position, you will need to reference it (as a static tf for example) to the position of each tag [...]

I would agree with that

> So If i would want to get one position out of a board of 4 markers, I have to deal with the 4 detected fiducial frames of this package, and combine them to a group estimate with other techniques. Is this what you intend?

yep

closed 2021-03-26 by fmros