UbiquityRobotics/fiducials #252: rosdep for noetic

tanyouliang95, 2021-06-23:
Run this on:
- ubuntu 20.04
- ros: noetic

When run:
```
sudo apt-get update && rosdep update && rosdep install --from-paths src --ignore-src -yr
```

```
E: Unable to locate package python-joblib
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y python-cairosvg] failed
  apt: command [sudo -H apt-get install -y python-joblib] failed
  apt: Failed to detect successful installation of [python-cairosvg]
  apt: Failed to detect successful installation of [python-joblib]
```

This is due to:
https://github.com/UbiquityRobotics/fiducials/blob/96b329a732f19d0f585c3740d42875a65168f31b/aruco_detect/package.xml#L30-L31

Which is using `python3-joblib` and `python-cairosvg` for noetic. Other than this, the package works fine.

rohbotics, 2021-07-19:
https://github.com/ros/rosdistro/pull/30241 and https://github.com/ros/rosdistro/pull/30170 solve this.

Thanks for reporting it.

closed 2021-07-19 by rohbotics

youliangtan, 2021-07-27:
Thanks for for adressing this. However, i believe you will still need to update the `package.xml` to:
```xml
 <depend>python3-cairosvg</depend> 
 <depend>python3-joblib</depend> 
```
Not sure whether will it interfere if using other ros distros.

rohbotics, 2021-07-27:
@youliangtan, those changes are now available in the noetic-devel branch.